duff2013/ulptool #3: ADC Example Will not compile

9H5G, 2018-06-04:
Great work and great examples - thank you.  All the examples work fine apart from the ADC which trowns the following compile error: 

python "/home/john/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp/esp32ulp_build_recipe.py" -b /tmp/arduino_build_531271 -p /home/john/Arduino/hardware/espressif/esp32 --DF_CPU=240000000L --DARDUINO=10805 --DARDUINO_=LOLIN32 --DARDUINO_ARCH_=ESP32 --DARDUINO_BOARD="LOLIN32" --DARDUINO_VARIANT="lolin32"
ULP Assembly File(s) Detected: ['adc.s']
/home/john/Arduino/hardware/espressif/esp32/tools/esp32ulp-elf-binutils/bin/esp32ulp-elf-ld -o ulp_main.elf -A elf32-esp32ulp -Map=ulp_main.map -T ulp_main.common.ld adc.ulp.o

exit status 1
Error compiling for board WEMOS LOLIN32.
Any thoughts?  Thanks!

duff2013, 2018-06-04:
Did you increase the ulp memory in sdkconfig? If not try raising it to 1024 by finding the macro 'CONFIG_ULP_COPROC_RESERVE_MEM'.

9H5G, 2018-06-04:
Thank you! RTFM always helps.... :)  BTW - running fine on Ubuntu 18.06

duff2013, 2018-06-04:
I'm glad you got it working, the error msgs are kind of cryptic which makes it hard to debug sometimes:(

9H5G, 2018-06-04:
Well, almost.  Compiles fine with memory extended to 1024 but on running, it returns this:  

Not ULP wakeup
ESP_ERROR_CHECK failed: esp_err_t 0x104 at 0x400d0b48
file: "/home/john/Arduino/ulp_examples/ulp_adc/ulp_adc.ino" line 52
func: void init_ulp_program()
expression: err

Backtrace: 0x4008771c:0x3ffca540 0x40087b65:0x3ffca560 0x400d0b48:0x3ffca580 0x400e4d7b:0x3ffca5a0

Rebooting...

Hmm...

duff2013, 2018-06-05:
Hmmm, I don't currently have esp Arduino setup on my machine but try increasing the memory size again, maybe 2048 bytes. I usually allocate memory in a power of 2's. But that error from what I read says you need more ulp memory.

duff2013, 2018-06-19:
Did you try increasing the memory yet? I'm going to close this issue tomorrow if I don't here back from you.

9H5G, 2018-06-19:
I worked around it by recoding. I didn’t need the temperature lookup table.
1024 was enough for my needs. Thank you.

duff2013, 2018-06-19:
Glad you got it to work!

closed 2018-06-19 by duff2013