duff2013/ulptool #5: Can't compile

ripper121, 2018-08-01:
I cant compile because of some python errors.

`Archiving built core (caching) in: C:\Users\sts\AppData\Local\Temp\arduino_cache_974869\core\core_espressif_esp32_esp32_PSRAM_disabled,PartitionScheme_default,FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_d446da940728032ed81daec0507ab040.a
Traceback (most recent call last):

  File "C:\Users\sts\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 440, in <module>

    main(sys.argv[1:])

  File "C:\Users\sts\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 139, in main

    build_ulp(options.bpath, options.ppath, ulp_files, board_options)

  File "C:\Users\sts\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 163, in build_ulp

    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)

  File "C:\Python27\lib\subprocess.py", line 710, in __init__

    errread, errwrite)

  File "C:\Python27\lib\subprocess.py", line 958, in _execute_child

    startupinfo)

WindowsError: [Error 2] Das System kann die angegebene Datei nicht finden

exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.`

duff2013, 2018-08-01:
What version of Windows are you using?

ripper121, 2018-08-01:
Windows 10 x64

ripper121, 2018-08-03:
> :
> So I don't have Windows 10 and it looks like I can't upgrade for free
> anymore, I have Windows 7 which I'm installing now in a virtual box. Are
> you able to help me with Windows 10?

please write me a mail to ripper121@gmail.com then i can help you.

duff2013, 2018-08-26:
Do you have the latest [ESP32-Arduino](https://github.com/espressif/arduino-esp32) installed? Then you need my latest [Arduino_ulp](https://github.com/duff2013/arduino_ulp).

ESP32 moved a compiling script into new place so my latest repository reflects that change but won't work with old installs of ESP32-Arduino

nzagorec, 2018-08-26:
Damn, I was using 1.0.0 (latest stable), not latest from master. Not it works, thanks ;)

closed 2019-01-03 by duff2013