duff2013/ulptool #6: Unsuccessful compiling

yankobogdan, 2018-08-16:
Arduino: 1.8.5 (Windows 7), Плата:"TTGO LoRa32-OLED V1, 80MHz, 921600, None"
Traceback (most recent call last):
  File "C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 440, in <module>
    main(sys.argv[1:])
  File "C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 139, in main
    build_ulp(options.bpath, options.ppath, ulp_files, board_options)
  File "C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 153, in build_ulp
    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.STDOUT,shell=False)
  File "C:\Python27\lib\subprocess.py", line 394, in init
    errread, errwrite)
  File "C:\Python27\lib\subprocess.py", line 644, in _execute_child
    startupinfo)
WindowsError: [Error 2] 
exit status 1

yankobogdan, 2018-08-16:
And still compiling issue on Win 10 and Mac OS High Sierra

duff2013, 2018-08-16:
Can we start with Windows 7 first? Turn on verbose compile in Arduino IDE preferences and copy and past the error using the example program from my [README.md](https://github.com/duff2013/arduino_ulp/blob/master/README.md). I need to see the build process along with the python error.

duff2013, 2018-08-16:
Also what version of python do you have?

yankobogdan, 2018-08-16:
Thanks for fast reply. Python ver. 2.7.15

Error code:
Arduino: 1.8.5 (Windows 7), Board: "TTGO LoRa32-OLED V1, 80MHz, 921600, None"
Archiving built core (caching) in: C:\Users\bogdan\AppData\Local\Temp\arduino_cache_206580\core\core_esp32_esp32_ttgo-lora32-v1_FlashFreq_80,UploadSpeed_921600,DebugLevel_none_0dd32befce4daf6467bf62049c4fab56.a
Traceback (most recent call last):
  File "C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 440, in <module>
    main(sys.argv[1:])
  File "C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 139, in main
    build_ulp(options.bpath, options.ppath, ulp_files, board_options)
  File "C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 153, in build_ulp
    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.STDOUT,shell=False)
  File "C:\Python27\lib\subprocess.py", line 394, in init
    errread, errwrite)
  File "C:\Python27\lib\subprocess.py", line 644, in _execute_child
    startupinfo)
WindowsError: [Error 2] 
exit status 1
Error compiling for board TTGO LoRa32-OLED V1.
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

duff2013, 2018-08-16:
Can you make sure this is checked, thanks.
[image]

yankobogdan, 2018-08-16:
Bogdan Yanko, [16.08.18 22:03]
/xtensa-esp32-elf-ar" cru  "C:\Users\bogdan\AppData\Local\Temp\arduino_build_131113/arduino.ar" "C:\Users\bogdan\AppData\Local\Temp\arduino_build_131113\core\main.cpp.o"
Archiving built core (caching) in: C:\Users\bogdan\AppData\Local\Temp\arduino_cache_206580\core\core_esp32_esp32_ttgo-lora32-v1_FlashFreq_80,UploadSpeed_921600,DebugLevel_none_0dd32befce4daf6467bf62049c4fab56.a
python "C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py" -b C:\Users\bogdan\AppData\Local\Temp\arduino_build_131113 -p C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0 --DF_CPU=240000000L --DARDUINO=10805 --DARDUINO_=TTGO_LoRa32_V1 --DARDUINO_ARCH_=ESP32 --DARDUINO_BOARD="TTGO_LoRa32_V1" --DARDUINO_VARIANT="ttgo-lora32-v1"
ULP Assembly File(s) Detected: ['ulp.s']
Traceback (most recent call last):
  File "C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 440, in <module>
    main(sys.argv[1:])
  File "C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 139, in main
    build_ulp(options.bpath, options.ppath, ulp_files, board_options)
  File "C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py", line 153, in build_ulp
    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.STDOUT,shell=False)
  File "C:\Python27\lib\subprocess.py", line 394, in init
    errread, errwrite)
  File "C:\Python27\lib\subprocess.py", line 644, in _execute_child
    startupinfo)
WindowsError: [Error 2] 
exit status 1
Error compiling for board TTGO LoRa32-OLED V1.

Bogdan Yanko, [16.08.18 22:03]
"-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/newlib" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/nvs_flash" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/openssl" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/spi_flash" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/sdmmc" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/spiffs" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/tcpip_adapter" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/vfs" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/wear_levelling" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/xtensa-debug-module" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/coap" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/console" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/expat" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/json" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/lwip" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/newlib" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/nghttp" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/soc" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_TTGO_LoRa32_V1 -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="TTGO_LoRa32_V1" -DARDUINO_VARIANT="ttgo-lora32-v1"  -DESP32 -DCORE_DEBUG_LEVEL=0 "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\variants\ttgo-lora32-v1" "C:\Users\bogdan\AppData\Local\Temp\arduino_build_131113\sketch\test_ulp.ino.cpp" -o "C:\Users\bogdan\AppData\Local\Temp\arduino_build_131113\sketch\test_ulp.ino.cpp.o"
C:\Users\bogdan\Desktop\test_ulp\test_ulp.ino: In function 'void init_run_ulp(uint32_t)':
C:\Users\bogdan\Desktop\test_ulp\test_ulp.ino:27:15: warning: variable 'err' set but not used [-Wunused-but-set-variable]
     esp_err_t err = ulp_load_binary(0, ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start) / sizeof(uint32_t));
               ^
Compiling libraries...
Compiling core...
Using previously compiled file: C:\Users\bogdan\AppData\Local\Temp\arduino_build_131113\core\esp32-hal-adc.c.o
[... cut ...] s\ttgo-lora32-v1" "C:\Users\bogdan\AppData\Local\Temp\arduino_build_131113\sketch\test_ulp.ino.cpp" -o "nul"
"C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/config" "-IC:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0

duff2013, 2018-08-16:
Can you compile a blank sketch?

```
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

duff2013, 2018-08-16:
Couple of things to check: 

Make sure you have [binutils toolchain](https://github.com/espressif/binutils-esp32ulp/releases) installed: 
**~\Documents\Arduino\Hardware\espressif\esp32\tools\esp32ulp-elf-binutils**

Is the esptool.exe located here? 
**~\Documents\Arduino\Hardware\espressif\esp32\tools\esptool**
If so then change the Platform.txt file line #6 to:
**tools.esptool.path={runtime.platform.path}/tools/esptool/**

yankobogdan, 2018-08-17:
My path to hardware folder looks like:
C:\Users\bogdan\AppData\Local\Arduino15\packages\esp32\hardware

yankobogdan, 2018-08-17:
binutils toolchain is installed
Mb you can archive your Arduino ide files to one folder and share?

duff2013, 2018-08-17:
How did you install esp32 into Arduino, through the Arduino Package Manager?

duff2013, 2018-08-17:
This video show how installed esp32 into Arduino, can you try to reinstall Arduino and esp32 this way?
https://www.youtube.com/watch?v=QBjiuOcJics

yankobogdan, 2018-08-17:
Solved!!!
The problem was that linker looking for esptool.exe in ...esp32/tools/ but it was inside esptool folder os I've just copy esptool.exe to tools folder and its compiled. My goal to try read i2c si7021 sensor, save date and some times wakeup main CPU.
Anyway thanks!

yankobogdan, 2018-08-17:
Solved!!!
The problem was that linker looking for esptool.exe in ...esp32/tools/ but it was inside esptool folder os I've just copy esptool.exe to tools folder and its compiled. My goal to try read i2c si7021 sensor, save the date and sometimes wakeup main CPU.
Anyway thanks!

duff2013, 2018-08-17:
Great, so I think espressif moved the esptool into it's own folder recently breaking my script, I'll update the script to reflect that change.

closed 2018-08-17 by duff2013