duff2013/ulptool #9: Minor issue for ulp I2C

Paulraita, 2018-09-11:
Hello, @duff2013.
Thanks for your great work. I encountered a minor issue for ulp I2C.
And I fixed this issue as I mentioned here.
https://github.com/espressif/arduino-esp32/issues/1854#issue-358893037
I hope this will be a contribution to make this library perfect.
I used Ubuntu 18.04 with Arduino IDE 1.8.6 and latest Arduino ESP32 core library (2018/8 updated).
Thanks.

duff2013, 2018-09-16:
I looked at this but not sure what change you made?

Paulraita, 2018-09-16:
There is no issue in compiling.
This kind of issue happens when I upload the sketch to the ESP32 board.
It shows ESP_ERR_INVALID_SIZE error but actually this is wrong result. Your library works well for small size .s files. I think there is some minor issue there. The user_ulp_load_binary() function is same as ulp_load_binary() function. I couldn't make it work with original library. So I tried to copy the codes for ulp_load_binary() and paste them in my project for debug. This is very strange but anyway, I could solve this ESP_ERR_INVALID_SIZE error in this way.
I am not sure if this is helpful.

closed 2018-10-14 by Paulraita

duff2013, 2018-10-18:
I'm still interested in this problem but I'm busy now and not currently working on any esp projects but when I come back to it I'll dig deeper into this.