duff2013/ulptool #11: Can't compile (undefined reference on libs)

dmitrij9992905, 2018-11-17:
I'm trying to compile one of examples and get errors:
`sketch\ulp.ino.cpp.o:(.literal._Z5setupv+0xc): undefined reference to `ulp_count'

sketch\ulp.ino.cpp.o:(.literal._Z5setupv+0x14): undefined reference to `_binary_ulp_main_bin_end'

sketch\ulp.ino.cpp.o:(.literal._Z5setupv+0x18): undefined reference to `_binary_ulp_main_bin_start'

sketch\ulp.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `ulp_entry'`

This observes with new platform.txt (v 1.0.0) file both on Windows and Linux (Windows 10 and Ubuntu Linux 16.04, python 2.7)

duff2013, 2018-11-18:
Did you install esp32 core using Arduino?

dmitrij9992905, 2018-11-18:
Yes, I installed esp32 core, other examples compiles fine. Only sketches
with ulp references gives these errors

dmitrij9992905, 2018-11-18:
Before installing these ulp lib and its toolchain it works with macros normally

duff2013, 2018-11-18:
There is issue when you install the esp32 core through Arduino Board Manger. If you can install the core this way -> [link windows](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md), [link linux](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md) it should work.

duff2013, 2018-11-21:
I think I fixed it, you can copy this updated platform.txt and paste it to your platform.txt.

dmitrij9992905, 2018-11-22:
[image]
Now it gives this error

duff2013, 2018-11-23:
Turn on "verbose for compilation" and post the full log.

dmitrij9992905, 2018-11-23:
[image]
A verbose log on the place that shows error

dmitrij9992905, 2018-11-23:
A full log
`

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\dmitr\AppData\Local\Arduino15\packages -hardware D:\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\dmitr\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\libraries -fqbn=espressif:esp32:esp32:PSRAM=disabled,PartitionScheme=default,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10807 -build-path C:\Users\dmitr\AppData\Local\Temp\arduino_build_279015 -warnings=all -build-cache C:\Users\dmitr\AppData\Local\Temp\arduino_cache_205823 -prefs=build.warn_data_percentage=75 -verbose D:\Arduino\ulp\ulp.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\dmitr\AppData\Local\Arduino15\packages -hardware D:\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\dmitr\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\libraries -fqbn=espressif:esp32:esp32:PSRAM=disabled,PartitionScheme=default,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10807 -build-path C:\Users\dmitr\AppData\Local\Temp\arduino_build_279015 -warnings=all -build-cache C:\Users\dmitr\AppData\Local\Temp\arduino_cache_205823 -prefs=build.warn_data_percentage=75 -verbose D:\Arduino\ulp\ulp.ino
Using board 'esp32' from platform in folder: D:\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: D:\Arduino\hardware\espressif\esp32
Detecting libraries used...
"D:\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/config" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/bluedroid" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/bluedroid/api" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/app_trace" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/app_update" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/bootloader_support" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/bt" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/driver" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp32" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_adc_cal" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_http_client" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_https_ota" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp-mqtt" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp-tls" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/ethernet" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/fatfs" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/freertos" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/heap" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/http_server" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/jsmn" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/log" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/mdns" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/mbedtls" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/mbedtls_port" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/newlib" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/nvs_flash" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/openssl" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/spi_flash" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/sdmmc" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/smartconfig_ack" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/spiffs" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/tcpip_adapter" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/tcp_transport" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/ulp" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/vfs" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/wear_levelling" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/xtensa-debug-module" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/lwip" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/coap" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/console" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/expat" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/json" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/newlib" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/nghttp" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/soc" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls - [... cut ...] " "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/soc" "-ID:\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/wpa_supplicant" -b "C:\\Users\\dmitr\\AppData\\Local\\Temp\\arduino_build_279015" -p "D:\\Arduino\\hardware\\espressif\\esp32" --DF_CPU=240000000L --DARDUINO=10807 --DARDUINO_=ESP32_DEV --DARDUINO_ARCH_=ESP32 "--DARDUINO_BOARD=\"ESP32_DEV\"" "--DARDUINO_VARIANT=\"esp32\""
exec: "python": executable file not found in %PATH%
Ошибка компиляции для платы ESP32 Dev Module.`

duff2013, 2018-12-06:
Your python is not working correctly i.e. -> exec: "python": executable file not found in %PATH%

duff2013, 2019-01-20:
This issue will be fixed, it's an issue with my Arduino_ulp code and v1.0.1.

closed 2019-01-20 by duff2013

Dserog, 2019-02-15:
@duff2013  I am having a similar problem with mac and board v1.0.1 . Has this been fixed as of yet? 
 Log below. 

/folders/fl/vbb1l0y56sj4__d74gd7df3m0000gn/T/arduino_build_84578/sketch/ulp_README.ino.cpp.o:(.literal._Z5setupv+0x10): undefined reference to `ulp_count'
/var/folders/fl/vbb1l0y56sj4__d74gd7df3m0000gn/T/arduino_build_84578/sketch/ulp_README.ino.cpp.o:(.literal._Z5setupv+0x18): undefined reference to `_binary_ulp_main_bin_end'
/var/folders/fl/vbb1l0y56sj4__d74gd7df3m0000gn/T/arduino_build_84578/sketch/ulp_README.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `_binary_ulp_main_bin_start'
/var/folders/fl/vbb1l0y56sj4__d74gd7df3m0000gn/T/arduino_build_84578/sketch/ulp_README.ino.cpp.o:(.literal._Z5setupv+0x20): undefined reference to `ulp_entry'
collect2: error: ld returned 1 exit status

duff2013, 2019-02-15:
Do you have latest version installed? Please open this as a new issue.

ArnieO, 2020-07-07:
```
undefined reference to `ulp_count'
undefined reference to `_binary_ulp_main_bin_end'
undefined reference to `_binary_ulp_main_bin_start'
undefined reference to `ulp_entry'
```

- Arduino IDE 1.8.13
- ESP32 boards installed as per https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md
- ulptool v2.4.2
- platform.local.txt in correct location (see below)

Please advise.

[image]