duff2013/ulptool #13: Change i2c_bitbang for MPU6050

Redstoned, 2019-01-13:
I wanted to change the i2c_bitbang example to get it running with the MPU6050 accelerometer/gyro.
Could You please help me with these values as I have no clue what they are (ulp_i2c_bitbang/i2c_dev.s):
[image]

duff2013, 2019-01-13:
I don't have any clue what those values are either except they look like registers of some kind. I haven't used that sketch and probably should get rid of it until someone gets it working.

Redstoned, 2019-01-13:
Is this the repo from which you have the example?
I would then ask there.
https://github.com/espressif/esp-iot-solution

duff2013, 2019-01-13:
I never used this example actually, I was able to compile it using the old method of using ulp assembly files but that was it. I don't have any equipment to test if it actually works thats why I might just get rid of it from the examples.