duff2013/ulptool #14: Installation not working under ESP32 1.0.1 (I'm fixing it)

unixbigot, 2019-01-13:
I'll have a PR for 101 hopefully tomorrow.  An a semi-automated installation script.

duff2013, 2019-01-13:
This is fine as long as it doesn't break the stable release from working ([v1.0.0](https://github.com/espressif/arduino-esp32/releases/tag/1.0.0)). Will this script work for any OS?

unixbigot, 2019-01-13:
Mac and linux are quite similar.   Windows will need a separate script.   Avoiding 'just copy over platform.txt and break any other modifications' will reduce breakage at version changes.

BTW 1.0.1 is the stable release now, it went gold last week.

MartinLocher, 2019-01-14:
Cool any update on this ?

s60sc, 2019-02-03:
I installed this library on my Windows 10, Python 3.7, Arduino 1.8.8, ESP 1.0.1 environment and it was straightforward to get working. 
`esp32ulp_build_recipe.py` needed a few minor syntax changes
The changes to `platform.txt` for ULP to work need to be copied into the 1.0.1 version of `platform.txt` 

[platform.txt](https://github.com/duff2013/ulptool/files/2825812/platform.txt)
[esp32ulp_build_recipe.py.txt](https://github.com/duff2013/ulptool/files/2825813/esp32ulp_build_recipe.py.txt)   

Thanks for a useful library

nsollars, 2019-06-23:
Is this fix good for 1.0.2?

duff2013, 2019-06-23:
1.0.2 should work fine now.

nsollars, 2019-06-23:
Ok, il undo everything thats there and rrapply everything from a fresh download and see what happens.

usmanshahid001, 2019-12-19:
hi i am using 1.0.4-rc2 .i followed all the steps of manual installation but problem is that when i compile the given examples it produce error that unable to find "ulptool.h".it is there in the include folder.i am unable to figure it out

2010019970909, 2020-11-13:
@usmanshahid001,

You were using Python 3, but this project only runs on Python 2 for the moment.

Vincent

yankobogdan, 2021-03-22:
> @usmanshahid001,
> 
> You were using Python 3, but this project only runs on Python 2 for the moment.
> 
> Vincent

Thanks, for your reply. Mb somebody can advise how I can choose Python 2 for compilation ulp connected stuff under Mac OS?
I have the same errors as @usmanshahid001