duff2013/ulptool #17: compile error ADC example

BuckfastBeekeeper, 2019-01-20:
But the compiler gives an error. I use 1.0.0 and 1.0.1 both. 
C:\Users\H\AppData\Local\Temp\arduino_build_268197\sketch\ulp_adc.ino.cpp.o:(.literal._Z5setupv+0xc): undefined reference to `_binary_ulp_main_bin_end' 
C:\Users\H\AppData\Local\Temp\arduino_build_268197\sketch\ulp_adc.ino.cpp.o:(.literal._Z5setupv+0x10): undefined reference to `_binary_ulp_main_bin_start' C:\Users\H\AppData\Local\Temp\arduino_build_268197\sketch\ulp_adc.ino.cpp.o:(.literal._Z5setupv+0x28): undefined reference to `ulp_low_threshold' 
C:\Users\H\AppData\Local\Temp\arduino_build_268197\sketch\ulp_adc.ino.cpp.o:(.literal._Z5setupv+0x2c): undefined reference to `ulp_high_threshold' 
C:\Users\H\AppData\Local\Temp\arduino_build_268197\sketch\ulp_adc.ino.cpp.o:(.literal._Z5setupv+0x3c): undefined reference to `ulp_sample_counter' 
C:\Users\H\AppData\Local\Temp\arduino_build_268197\sketch\ulp_adc.ino.cpp.o:(.literal._Z5setupv+0x48): undefined reference to `ulp_ADC_reading' 
C:\Users\H\AppData\Local\Temp\arduino_build_268197\sketch\ulp_adc.ino.cpp.o:(.literal._Z5setupv+0x50): undefined reference to `ulp_entry' 

collect2.exe: error: ld returned 1 exit status Bibliotheek ESP32 op versie 1.0 in map: 
C:\Users\Hennie\Desktop\arduino-1.8.7\portable\packages\esp32\hardware\esp32\1.0.0\libraries\ESP32 
wordt gebruikt exit status 1 Fout bij het compileren voor board TTGO LoRa32-OLED V1

BuckfastBeekeeper, 2019-01-20:
"Did you install esp32 core through the board manager? "
yes..
"Currently only v1.0.0 is supported."
 is installed
The ReadMe example gives this error:
C:\Users\H\Desktop\arduino-1.8.7\portable\sketchbook\arduino_ulp-master\ulp_examples\ulp_README\README\README.ino: In function 'void init_run_ulp(uint32_t)':

README:25:5: error: 'declare' was not declared in this scope

     declare ulp_count;

     ^

exit status 1
'declare' was not declared in this scope

duff2013, 2019-01-20:
```declare ulp_count``` I'm not sure where you are getting this from? Did you edit the sketch? Also where is your esp32 core installed, it should be ```C:\Users\<USERNAME>\AppData\Local\Arduino15\packages\esp32```

guteSeite, 2019-01-29:
Sorry, I have compiling error too with the readme example, using 1.8.7 / 1.0.0
e:\Temp\arduino_build_256402\sketch\sketch_jan29a.ino.cpp.o:(.literal._Z5setupv+0xc): undefined reference to `ulp_count'

e:\Temp\arduino_build_256402\sketch\sketch_jan29a.ino.cpp.o:(.literal._Z5setupv+0x14): undefined reference to `_binary_ulp_main_bin_end'

e:\Temp\arduino_build_256402\sketch\sketch_jan29a.ino.cpp.o:(.literal._Z5setupv+0x18): undefined reference to `_binary_ulp_main_bin_start'

e:\Temp\arduino_build_256402\sketch\sketch_jan29a.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `ulp_entry'

collect2.exe: error: ld returned 1 exit status

Any idea?

guteSeite, 2019-02-07:
this is my error with the new ulptool-2.1.0. :
xtensa-esp32-elf-gcc: error: c:\Temp\arduino_build_58216/sketch/ulp_main.bin.bin.o: No such file or directory

2010019970909, 2019-02-09:
Hello @guteSeite,

> Sorry, I have compiling error too with the readme example, using 1.8.7 / 1.0.0
> e:\Temp\arduino_build_256402\sketch\sketch_jan29a.ino.cpp.o:(.literal._Z5setupv+0xc): undefined reference to [...]

I manage to get the same error by removing the 'platform.local.txt' file from the platform folder (see setup in readme). So it can be that, check if this file is in the good folder and let us know.

I get this error without the file for the ulp_README sketch (it's exactly the same error):

```
C:\Users\Vincent\AppData\Local\Temp\arduino_build_420701\sketch\ulp_README.ino.cpp.o:(.literal._Z5setupv+0x10): undefined reference to `ulp_count'

C:\Users\Vincent\AppData\Local\Temp\arduino_build_420701\sketch\ulp_README.ino.cpp.o:(.literal._Z5setupv+0x18): undefined reference to `_binary_ulp_main_bin_end'

C:\Users\Vincent\AppData\Local\Temp\arduino_build_420701\sketch\ulp_README.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `_binary_ulp_main_bin_start'

C:\Users\Vincent\AppData\Local\Temp\arduino_build_420701\sketch\ulp_README.ino.cpp.o:(.literal._Z5setupv+0x20): undefined reference to `ulp_entry'

collect2.exe: error: ld returned 1 exit status

exit status 1
```

So under Windows from the path `C:\Users\<USERNAME>\AppData\Local\Arduino15\packages\esp32`, you should have this tree:

```
path/
├───hardware
│   └───esp32
│       └───1.0.1
│           ├───cores
│           ├───libraries
│           ├───tools
│           ├───variants
│           └─── *'platform.local.txt'*
└───tools
    ├───esptool_py
    │   └─── [...]
    ├───mkspiffs
    │   └─── [...]
    ├───*ulptool*
    │   └───*src*
    │       ├───*esp32*
    │       ├───*esp32ulp-elf-binutils*
    │       └───*ld*
    └───xtensa-esp32-elf-gcc
        └─── [...]
```