duff2013/ulptool #18: Can't compile ulp_adc - (bin/esp32ulp-elf-ld: region `ram' overflowed by 88 bytes)

2010019970909, 2019-01-20:
Hello,

I'm working on the Arduino 1.8.5 on Windows 10 (version 10.0.17763.253), with the core esp32 version 1.0.0. I've modified the two python's scripts to make them run under python 3.7 ([py3.7.zip](https://github.com/duff2013/arduino_ulp/files/2777061/py3.7.zip)). The _README.ino_ example works. But when I try to compile the _ulp_adc_ example I get this error:

```
ULP Assembly File(s) Detected: adc.s

C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\tools\binutils\esp32ulp-elf-binutils/bin/esp32ulp-elf-ld -o ulp_main.elf -A elf32-esp32ulp -Map=ulp_main.map -T ulp_main.common.ld adc.ulp.o

C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\tools\binutils\esp32ulp-elf-binutils/bin/esp32ulp-elf-ld: ulp_main.elf section '.data' will not fit in region 'ram'

C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\tools\binutils\esp32ulp-elf-binutils/bin/esp32ulp-elf-ld: region 'ram' overflowed by 88 bytes

exit status 1
Erreur de compilation pour la carte ESP32 Dev Module
```

So `'ram' overflowed by 88 bytes` even if I change `CONFIG_ULP_COPROC_RESERVE_MEM` to 2048

It comes from gen_binutils_ld_cmd() at first look, but I don't know what is exactly the issue.

Thanks in advance for your answer,
Vincent

duff2013, 2019-01-21:
Not sure why its blowing up the size, it compiles fine with 2.7. Python just calls binutil commands it doesn't compile anything so I don't think it has to do with python well except for the esp manpgen script but your getting the error well before it gets there.

2010019970909, 2019-01-21:
Yes it's weird. I only use my python 3.7 script so it's hard to know if it comes from my modifications in the originals scripts or if it's more an error which is due to the espressif utilities... Since last night I've added some modifications like directly printing the executed command in the terminal, displaying the version of python and displaying the core version ([scripts_py3.7.zip](https://github.com/duff2013/arduino_ulp/files/2778120/scripts_py3.7.zip)). Thus I have the last working commands, it could help for debugging:

```
C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -MMD -MP -DWITH_POSIX -DMBEDTLS_CONFIG_FILE=mbedtls\esp_config.h -DHAVE_CONFIG_H -MT adc.ulp.o -E -P -xc -o adc.ulp.pS -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/config -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/bluedroid -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/bluedroid/api -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/app_trace -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/app_update -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/bootloader_support -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/bt -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/driver -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/esp32 -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/esp_adc_cal -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/esp_http_client -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/esp-tls -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ethernet -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/fatfs -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/freertos -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/heap -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/jsmn -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/log -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/mdns -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/mbedtls -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/mbedtls_port -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/newlib -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/nvs_flash -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/openssl -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/spi_flash -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/sdmmc -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/smartconfig_ack -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/spiffs -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/tcpip_adapter -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/ulp -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/vfs -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/wear_levelling -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/xtensa-debug-module -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/coap -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/console -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/expat -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/json -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/lwip -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/newlib -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/nghttp -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/soc -I C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/include/wpa_supplicant -DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_=ESP32_DEV -DARDUINO_ARCH_=ESP32 -DARDUINO_BOARD="ESP32_DEV" -DARDUINO_VARIANT="esp32" -I C:\Users\Vincent\AppData\Local\Temp\arduino_build_753332\sketch -D__ASSEMB [... cut ...] P32 -DARDUINO_BOARD="ESP32_DEV" -DARDUINO_VARIANT="esp32" -I C:\Users\Vincent\AppData\Local\Temp\arduino_build_753332\sketch -D__ASSEMBLER__ C:\Users\Vincent\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\tools\sdk\include\ulp\ld\esp32.ulp.ld
```
Can you try my scripts with python 2.7 to see if it's compatible with this version? Do you think this error can come from the previous commands? It would be great to have a script running under python3 and python2.

Have a nice day,
Vincent

2010019970909, 2019-01-21:
Now, via PowerShell, I've tried to launch the script for Readme with python 2.7 (it works) and for the ulp_adc the script seems to work, but I get the same error.

duff2013, 2019-02-03:
There might be an issue with adding memory and how the esp32 checks at runtime I'll update this if find anything out.

duff2013, 2019-02-06:
This is a problem that I'm addressing now and it won't be an easy fix :(

duff2013, 2019-02-18:
The latest release should fix memory allocation problems.

2010019970909, 2019-02-19:
It works for Windows with Python 2. I had to add `compiler.s.cmd.windows=py -2 "{tools.ulptool.path}{tools.ulptool.cmd}"` in "platform.local.txt" since it's not compatible with Python 3. Now it works like a charm. Thanks a lot @duff2013.

closed 2019-02-19 by 2010019970909

rkit1, 2020-11-29:
This issue seems to be caused by having long file paths on windows. Moving the project closer to drive root solved it for me.

jbrepogmailcom, 2021-02-25:
Hello, I have same issue on Linux, can you advise how to modify platform.local.txt?

klugd, 2021-09-20:
On Linux solves me as I've increased 

#define CONFIG_ULP_COPROC_RESERVE_MEM 512 

to higher value in 

~/.arduio15/packages/esp32/hardware/esp32/1.0.5/tools/sdk/include/sdkconfig.h