duff2013/ulptool #20: "python": executable file not found in %PATH%

chrs37, 2019-01-22:
I've just followed the install instructions and are trying to get a example sketch compiled, but keep getting this error: exec: "python": executable file not found in %PATH%

I'm using version 1.0.0 of the programmer esp32, installed via the board manager. I also downloaded python 2.7

Any idea what is wrong?

-Christian

duff2013, 2019-01-23:
Something is wrong with Arduino trying to find your python path, but you need to provide more info like  what OS you have what version of Arduino and such. Also what happens when you compile a blank sketch.

chrs37, 2019-01-23:
I'm using windows 10 and arduino 1.8.8.  Compiling a blank sketch gives the same error message. I've installed python in C:\Python27

duff2013, 2019-01-23:
Can you open your cmd prompt and type **python**, what happens? Do you get into python shell?

chrs37, 2019-01-23:
After setting the dir, this happens. Is this the python shell?
[image]

chrs37, 2019-01-24:
I used the windows installer to install arduino and downloaded the esp32 core via the board manager. This is what happens when I enable verbose output:

`python "C:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/config" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bluedroid" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bluedroid/api" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/app_trace" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/app_update" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bootloader_support" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bt" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/driver" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp32" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp_http_client" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp-tls" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/ethernet" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/fatfs" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/freertos" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/heap" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/jsmn" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/log" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/mdns" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/mbedtls" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/mbedtls_port" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/newlib" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/nvs_flash" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/openssl" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/spi_flash" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/sdmmc" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/smartconfig_ack" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/spiffs" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/tcpip_adapter" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/ulp" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/vfs" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/wear_levelling" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/xtensa-debug-module" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/coap" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/console" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/expat" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/json" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/lwip" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/newlib" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/nghttp" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/soc" "-IC:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/wpa_supplicant" -b "C:\\Users\\peppe\\AppData\\Local\\Temp\\arduino_build_513812" -p "C:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0" -u "C:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\binutils\\esp32ulp-elf-binutils/bin/" -x "C:\\Users\\peppe\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/" --DF_CPU=240000000L --DARDUINO=10808 --DARDUINO_=FEATHER_ESP32 --DARDUINO_ARCH_=ESP32 "--DARDUINO_BOARD=\"FEATHER_ESP32\"" "--DARDUINO_VARIANT=\"feather_esp32\""
`

duff2013, 2019-01-25:
where is the error message in what you just posted? Also [here](https://user-images.githubusercontent.com/13698937/51620076-ebb6e000-1f31-11e9-9d74-9fb7f97734a0.PNG) you are in the python directory can you run python in other directories?

chrs37, 2019-01-25:
The error message ("python": executable file not found in %PATH%) is in the line after the verbose output I just posted. I can only acces python from the python directory. I can run the command from the previous comment in the terminal, but only directly from the python directory.

chrs37, 2019-01-25:
I got the compiling working! I had to add Python to the PATH table (https://www.pythoncentral.io/add-python-to-path-python-is-not-recognized-as-an-internal-or-external-command/) and now the sketch compiles. I'll have to check later if the uploading also works

duff2013, 2019-01-25:
Yep thats where I was going, glad it works now.

closed 2019-01-25 by duff2013

linehammer, 2019-07-16:
import sys
if "C:\\My_Python_Lib" not in sys.path:
    sys.path.append("C:\\My_Python_Lib")

More on...[Python Path](http://net-informations.com/python/intro/path.htm)