duff2013/ulptool #21: Error when compiling large programs despite CONFIG_ULP_COPROC_RESERVE_MEM=8192

Redstoned, 2019-01-26:
Hi,
I just ran into the issue that when im writing one more command, the ulp program will not compile.
I have about 120 lines of commands without the jumppoint-definitions so around 480 bytes. As in the title mentioned, I set the memory to 8192 bytes. I have no clue what is going on.

I get the 'unable to cache built core'-message with any ulp-program but it still uploads.

Unable to cache built core, please tell esp32 maintainers to follow http://goo.gl/QdCUjo
python "C:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/ulp/esp32ulp_build_recipe.py" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/config" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bluedroid" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bluedroid/api" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/app_trace" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/app_update" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bootloader_support" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bt" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/driver" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp32" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp_http_client" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp-tls" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/ethernet" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/fatfs" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/freertos" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/heap" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/jsmn" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/log" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/mdns" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/mbedtls" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/mbedtls_port" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/newlib" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/nvs_flash" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/openssl" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/spi_flash" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/sdmmc" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/smartconfig_ack" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/spiffs" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/tcpip_adapter" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/ulp" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/vfs" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/wear_levelling" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/xtensa-debug-module" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/coap" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/console" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/expat" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/json" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/lwip" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/newlib" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/nghttp" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/soc" "-IC:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/wpa_supplicant" -b "C:\\Users\\cedric\\AppData\\Local\\Temp\\arduino_build_500352" -p "C:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0" -u "C:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\binutils\\esp32ulp-elf-binutils/bin/" -x "C:\\Users\\cedric\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/" --DF_CPU=240000000L --DARDUINO=10808 --DARDUINO_=ESP32_DEV --DARDUINO_ARCH_=ESP32 "--DARDUINO_BOARD=\"ESP32_DEV\"" "--DARDUINO_VARIANT=\"doitESP32devkitV1\""
ULP Assembly File(s) Detected: ulp.s

C:\Users\cedric\AppData\Local\Arduino15\packages\esp32\tools\binutils\esp32ulp-elf-binutils/bin/esp32ulp-elf-ld -o ulp_main.elf -A elf32-esp32ulp -Map=ulp_main.map -T ulp_main.common.ld ulp.ulp.o

exit status 1
Fehler beim Kompilieren für das Board DOIT ESP32 DEVKIT V1.

Redstoned, 2019-01-27:
[size_test.zip](https://github.com/duff2013/arduino_ulp/files/2800101/size_test.zip)

The test program I wrote tests the limits.
I had the idea of changing the Freertos task size for Arduino to something higher as it helped me out in the past but it didn't work.
I'm talking about this file: C:\Users\cedric\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\main.cpp

Redstoned, 2019-01-28:
As discussed on the esp32 forum here are my results:
ULP programs larger than 512 bytes/ 125 commands won't run even if you set menuconfig properly.
I fixed line 7 in esp32/1.0.0/tools/sdk/include/ulp/ld/esp32.ulp.ld
from 
`    ram(RW) : ORIGIN = 0, LENGTH = CONFIG_ULP_COPROC_RESERVE_MEM`
to
`    ram(RW) : ORIGIN = 0, LENGTH = 8192`

and copied the ulp_load_binary-function from [here](https://github.com/espressif/esp-idf/blob/master/components/ulp/ulp.c)  in a new tab in Arduino called "ulp.c"
deleted the error return lines and renamed the function to unsafe_ulp_load_binary.
Then I referenced "ulp.c" and called the unsafe function instead of the safe one.

duff2013, 2019-01-30:
Not sure what is going on here but the RTC_SLOW_MEM is 8k. If you want to be brave install the esp-idf and try your ulp program there and see if it works.

duff2013, 2019-02-03:
I see now that ulp.c has a check for CONFIG_ULP_COPROC_RESERVE_MEM so maybe there is an issue there so I'll check that out if its pulling the precompiled one or the edited one I don't know yet.

duff2013, 2019-02-06:
This is a bigger problem than just what you proposed and fixed, there are linker scripts that need to be edited. Also they use a precompiled library for the ulp support hard coded in the build so I'm not sure how to handle it yet.

duff2013, 2019-02-08:
@Redstoned thanks for bringing this to my attention and the **size_test** sketch really helped me figure out what was going on. So come to find out Arduino core esp linker script is a produced when they build all the components so everything is fixed when Arduino uses it. I now have Arduino run a c preprocessor over it to bring in any changes made to sdkconfig.h at compile time. This is actually what I think they do when you use the esp-idf so now any changes to the sdkconfig.h will show up in the linker script. 

Another thing I had to do is add in a **ulptool** specific function to load the binary called ```ulptool_load_binary``` which will grab the correct memory size you allocate in the sdkconfig.h file. One thing to note is that linker script that the Arduino core esp uses only allows for 4096 bytes of RTC Slow Memory and I have no understanding why yet. 

I'll update this fix in the next couple of days.

Redstoned, 2019-02-10:
I've got even ULP programs that use 8192 bytes ulpoaded and working.
Here is my test sketch:
(Instructions are at the top of the ulp.s file)
[size_test.zip](https://github.com/duff2013/ulptool/files/2848633/size_test.zip)

duff2013, 2019-02-13:
I think the 4096 bytes is because you can reserve RTC Slow memory from c and cpp means which you could walk all over your ulp memory if the linker is not setup right or vice versa. I'm not going any farther now on this because I have fix for the linker script that esp core uses anyways. I'll be updating the ulptool in the next week for this.

duff2013, 2019-02-18:
The latest release should fix the memory allocation problem.