duff2013/ulptool #22: Please add that its working like a charm in version 1.0 of ESP32

MartinLocher, 2019-01-28:
Hello Sir,

with 1.0.1 this is not working, i saw that the comment with version 1.0 was removed, but still the issue persist.
It'll help people to adjust properly.

Thanks for your great work !!!
Best, Martin

duff2013, 2019-01-29:
Yes version 1.0.1 does not currently work but I have version that does but there are few other changes that I'm waiting to do first.

MartinLocher, 2019-02-02:
Hello Sir, so please add this to the read.me as people will get frustrated :-)

duff2013, 2019-02-03:
I'm working on a version the can use both, hopefully by tomorrow I will have it pushed.

duff2013, 2019-02-05:
There is new version that compiles both versions now, check it out let me know if you have problems. The install is a little different now so follow the new README