duff2013/ulptool #23: Issue arduino.h no such file or directory

TheIronMechanic, 2019-01-30:
Hi
After following the get started guide i experience following issue.
Every time I want to verify or upload a sketch to any esp32 board, I get an issue that looks like this:

- from your example sketch:
`Build options changed, rebuilding all
sketch\ulpexample.ino.cpp:1:21: fatal error: Arduino.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board NodeMCU-32S.
`
- or from an other sketch, no using ulp:
`Build options changed, rebuilding all
sketch\BLE_scan.ino.cpp:1:21: fatal error: Arduino.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board NodeMCU-32S.`

but uploading to any other board type (like arduino uno or mega) works.

does anybody know how to fix it?

Used exactly your get started, with the board definition esp32 v1.0.0 in the arduino ide 1.8.5 on windows 10.

duff2013, 2019-01-30:
I think you need to reinstall esp32 core using the board manager and before you do any ulp stuff compile a blank sketch to make sure it works. Not sure how Arduino.h went missing on your system but it esp build can't find it.

TheIronMechanic, 2019-02-02:
I completely removed and then reinstalled the esp32 core.
tested it everything works.
made the changes to ult, and I get different errors this time.

`
Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\Lucas\AppData\Local\Temp\arduino_cache_562248\core\core_esp32_esp32_nodemcu-32s_FlashFreq_80,UploadSpeed_921600_165aa9b4f273454066eb068852ab8e83.a
exec: "python": executable file not found in %PATH%
Error compiling for board NodeMCU-32S.
`
But I'm at a different error now. Hurray!

duff2013, 2019-02-03:
You need to set your python PATH for windows.

TheIronMechanic, 2019-02-03:
OMG I totally forgot about it
Thanks a lof for your quick help!

closed 2019-02-03 by TheIronMechanic