duff2013/ulptool #24: Fails on linking

dgleeson1000, 2019-01-30:


dgleeson1000, 2019-01-30:
Arduino: 1.8.7 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Default, QIO, 80MHz, 4MB (32Mb), 921600, None"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dave/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dave/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dave/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10807 -build-path /var/folders/zn/jz6l7fxd78s9fk958hvmv2lr0000gn/T/arduino_build_525802 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/dave/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/dave/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/dave/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/Users/dave/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool.path=/Users/dave/Library/Arduino15/packages/esp32/tools/esptool/2.3.1 -prefs=runtime.tools.esptool-2.3.1.path=/Users/dave/Library/Arduino15/packages/esp32/tools/esptool/2.3.1 -verbose /Users/dave/Documents/Arduino/ulp_examples/ulp_adc/ulp_adc.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dave/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dave/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dave/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10807 -build-path /var/folders/zn/jz6l7fxd78s9fk958hvmv2lr0000gn/T/arduino_build_525802 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/dave/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/dave/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/dave/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/Users/dave/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool.path=/Users/dave/Library/Arduino15/packages/esp32/tools/esptool/2.3.1 -prefs=runtime.tools.esptool-2.3.1.path=/Users/dave/Library/Arduino15/packages/esp32/tools/esptool/2.3.1 -verbose /Users/dave/Documents/Arduino/ulp_examples/ulp_adc/ulp_adc.ino
Using board 'esp32' from platform in folder: /Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0
Using core 'esp32' from platform in folder: /Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0
Detecting libraries used...
/Users/dave/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/config -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/bluedroid -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/bluedroid/api -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/app_trace -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/app_update -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/bootloader_support -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/bt -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/driver -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/esp32 -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/esp_adc_cal -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/esp_http_client -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/esp-tls -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/ethernet -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/fatfs -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/freertos -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/heap -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/jsmn -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/log -I/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/ [... cut ...] rt.c.o
/Users/dave/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-ar cru /var/folders/zn/jz6l7fxd78s9fk958hvmv2lr0000gn/T/arduino_build_525802/arduino.ar /var/folders/zn/jz6l7fxd78s9fk958hvmv2lr0000gn/T/arduino_build_525802/core/libb64/cdecode.c.o
/Users/dave/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-ar cru /var/folders/zn/jz6l7fxd78s9fk958hvmv2lr0000gn/T/arduino_build_52580

duff2013, 2019-01-30:
did you increase the ulp memory size in [sdkconfig.h](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/config/sdkconfig.h#L89)?

duff2013, 2019-01-30:
Also does the README example work?

dgleeson1000, 2019-01-30:
Yes the README is working fine. I should have said that.

I cant find the file esp32/tools/sdk/include/sdkconfig.h 

I found
/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/sdkconfig
and I edited that and changed the line
CONFIG_ULP_COPROC_RESERVE_MEM=512
to
CONFIG_ULP_COPROC_RESERVE_MEM=2048

but I'm still getting the same error.
Cheers.

dgleeson1000, 2019-01-30:
Success..
I found sdkconfig.h in
/Users/dave/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/include/config
made the change and it's working fine now.

Thanks for all your help.

duff2013, 2019-01-30:
Yes it's **sdkconfig.h** going to close this issue now, glad it works!

closed 2019-01-30 by duff2013

duff2013, 2019-01-30:
Thats the runtime error, you will have dig for the reason I don't know.

dgleeson1000, 2019-01-31:
No worries. Thanks again fr all your help.