duff2013/ulptool #26: Compile fails for non-ulp project

Rain92, 2019-02-07:
Thanks to #25 I got this tool running nicely under windows python 3 and esp 1.0.1.
However every time I want to compile a project without any ulp code I get the error message:
xtensa-esp32-elf-gcc*: error: [...]\ulp_main.bin.bin.o: No such file or directory

It would be nice if this could be handled automatically.

duff2013, 2019-02-07:
oh man thanks for letting know!

2010019970909, 2019-02-07:
Yes, same error (I've used 'blink' to test it). But I don't think that comes from the 'esp32ulp_build_recipe.py' script since it just open, close 'ulp_main.ld' and exit the script in this case. My only clue is the 'platform.local.txt' file but I don't know the syntax...

duff2013, 2019-02-07:
For the time being to get it to compile just add blank .s file to your sketch.

2010019970909, 2019-02-07:
Examples are in a read only directory. I'll try to find an other workaround since just automatically creating an empty 'ulp.s' file is not really an option for the examples.

2010019970909, 2019-02-07:
After 'some' investigations, I found that the issue comes from the

`compiler.c.elf.extra_flags="-L{build.path}/sketch/" -T ulp_main.ld "{build.path}/sketch/ulp_main.bin.bin.o"`

in "platform.local.txt".

So it will be useful to add a conditional statement somehow... IDK for the moment.

Edit: I omit to say that commenting this line (Add ## in front of the line) is a tested working workaround for the examples.

duff2013, 2019-02-08:
I have a fix, basically I just create a blank temporary .s file and compile it to get those files after I delete that temporary file. This is not a complete fix because it might use 12 bytes Slow RTC memory but I hope it will work for now until I can figure out a better way. 

@2010019970909 The only way I can see the script being able to handle this is if there are no ulp files detected then delete the flags in my platform.local.txt file or add them to it if there are. But for now the fix I have works and need to publish some more pressing fixes for other big issues with how the ulp is allocated RTC Memory in the esp core at compile time.

2010019970909, 2019-02-08:
On my fork, I've added an modified 'platform.local.txt' file to launch a script ('recipe_c_combine_pattern.py') which build the correct command for the gcc compiler. As usually, I've tested it only on windows.