duff2013/ulptool #27: undefined reference

guteSeite, 2019-02-08:
old and new version same errors :-(

C:\Users\BG\AppData\Local\Temp\arduino_build_501687\sketch\sketch_feb08a.ino.cpp.o:(.literal._Z5setupv+0x10): undefined reference to `ulp_count'

C:\Users\BG\AppData\Local\Temp\arduino_build_501687\sketch\sketch_feb08a.ino.cpp.o:(.literal._Z5setupv+0x18): undefined reference to `_binary_ulp_main_bin_end'

C:\Users\BG\AppData\Local\Temp\arduino_build_501687\sketch\sketch_feb08a.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `_binary_ulp_main_bin_start'

C:\Users\BG\AppData\Local\Temp\arduino_build_501687\sketch\sketch_feb08a.ino.cpp.o:(.literal._Z5setupv+0x20): undefined reference to `ulp_entry'

collect2.exe: error: ld returned 1 exit status

duff2013, 2019-02-08:
Can you turn on **Show verbose output during** compilation  in the Arduino prefs and copy from where it says **ULP Assembly File(s) Detected:** down to the end where the error message shows up. I'm assuming you are using the ulp_README.ino example if not please do.

guteSeite, 2019-02-08:
Now I got it!
There was the error message: No ULP Assembly File(s) Detected...
Therefore the ulp.s file in my example folder arduino/libaries/ulp was not automatically detected.
I added the file manually and everything is fine now!!!
Sorry for the trouble.

duff2013, 2019-02-08:
Oh no you got the no ulp file bug:( I just committed a fix for that which you can download now.

Jayce89757, 2020-05-11:
@guteSeite Hi friend,
Do you solve the problem?
Is move ulp.s file to the specified location?
And where the location?