duff2013/ulptool #28: Installation is not working for mac on board v1.0.1

Dserog, 2019-02-15:
Yes, I tried the latest version on multiple machines and I am getting the same error. 

Log: 
/folders/fl/vbb1l0y56sj4__d74gd7df3m0000gn/T/arduino_build_84578/sketch/ulp_README.ino.cpp.o:(.literal._Z5setupv+0x10): undefined reference to `ulp_count' /var/folders/fl/vbb1l0y56sj4__d74gd7df3m0000gn/T/arduino_build_84578/sketch/ulp_README.ino.cpp.o:(.literal._Z5setupv+0x18): undefined reference to `_binary_ulp_main_bin_end'
/var/folders/fl/vbb1l0y56sj4__d74gd7df3m0000gn/T/arduino_build_84578/sketch/ulp_README.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `_binary_ulp_main_bin_start' /var/folders/fl/vbb1l0y56sj4__d74gd7df3m0000gn/T/arduino_build_84578/sketch/ulp_README.ino.cpp.o:(.literal._Z5setupv+0x20): undefined reference to `ulp_entry'
collect2: error: ld returned 1 exit status

Do you have latest version installed? Please open this as a new issue.

_Originally posted by @duff2013 in https://github.com/duff2013/ulptool/issues/11#issuecomment-463905620_

duff2013, 2019-02-15:
can you post the error from where it says **ULP Assembly File(s) Detected:**

Dserog, 2019-02-15:
@duff2013 it never says that.

Dserog, 2019-02-15:
Im pretty sure it's my file setup I've attached a screenshot. See anything wrong? 
[image]

duff2013, 2019-02-15:
The **ulp_examples** folder should be in ~/Documents/Arduino/

Also can you show me where you installed the **ulptool** folder, **binutils** folder and where you moved the **platform.local.txt** file?

Dserog, 2019-02-15:
@duff2013 I got it, for some reason not yet understood my compiler was looking at where I store my sketches, not my core files. So, I put the programmer.local.text file in ..documents/arduino/hardware/expressif/esp32 files as well as ..Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/ and the Readme file compiled.

duff2013, 2019-02-15:
The preferred way to install esp32 into Arduino is with the package manager which installs the esp stuff in ~/Library/Arduino15/packages/esp32/

But it looks like you installed it the manual way also in which the files would be in ~/Documents/Arduino/hardware/esp32/

It looks like the package manager is what Arduino is defaulting to. For future installs use the package manager thats what ulptool is designed to use. You can delete the ~/Documents/Arduino/hardware/esp32 if you want.

closed 2019-02-15 by Dserog