duff2013/ulptool #31: Arduino can not find ulptool.h

robouden, 2019-02-27:
I followed the directions of installing the tools, but I can not get it to work. Anyone on OSX has it working?

regards,
rob oudendijk

duff2013, 2019-02-27:
you downloaded the latest [release](https://github.com/duff2013/ulptool/releases/tag/2.3.0)?

robouden, 2019-02-27:
Duff,

Thanks for the fast reply.  Yes, downloaded and installed according to the
direction given on GitHub. But Arduino can not find the up-tool.h.

Any more suggestions?

regards
rob

Regards,
Rob Oudendijk Yuka Hayashi
http://yr-design.biz   http://oudendijk.biz  http://about.me/robouden
tel +81 80-22605966  Skype: robouden  Facebook:robouden
<http://on.fb.me/QeKw2P> linkedin:robouden
<https://www.linkedin.com/in/roboudendijk>

duff2013, 2019-02-27:
Using [ulp_README.ino](https://github.com/duff2013/ulptool/tree/master/src/ulp_examples/ulp_README) example can you turn on verbose compiling in Arduino and post from where it say how many ulp file detected.

robouden, 2019-02-27:
Duff,

Arduino cannot even find ulptool.h. How can it run the  .imo file?

regards
rob

Regards,
Rob Oudendijk Yuka Hayashi
http://yr-design.biz   http://oudendijk.biz  http://about.me/robouden
tel +81 80-22605966  Skype: robouden  Facebook:robouden
<http://on.fb.me/QeKw2P> linkedin:robouden
<https://www.linkedin.com/in/roboudendijk>

duff2013, 2019-02-27:
Can you post a screen shot of your directory where you installed the ulptool, with what you gave me I have no idea whats going on so I'm going to need much more info to help.

robouden, 2019-02-27:
Duff,

Hereby the folder structure.

regards
rob

i4things, 2019-03-03:
same issue here (help kindly appreciated) : 

C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\hardware -hardware C:\Users\admin\AppData\Local\Arduino15\packages -hardware C:\Users\admin\Documents\Arduino\hardware -tools C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\tools-builder -tools C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\hardware\tools\avr -tools C:\Users\admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\libraries -libraries C:\Users\admin\Documents\Arduino\libraries -fqbn=espressif:esp32:heltec_wifi_lora_32:FlashFreq=80,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\admin\AppData\Local\Temp\arduino_build_854219 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\admin\Documents\Arduino\sketch_mar03a\sketch_mar03a.ino
C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\arduino-builder -compile -logger=machine -hardware C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\hardware -hardware C:\Users\admin\AppData\Local\Arduino15\packages -hardware C:\Users\admin\Documents\Arduino\hardware -tools C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\tools-builder -tools C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\hardware\tools\avr -tools C:\Users\admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\libraries -libraries C:\Users\admin\Documents\Arduino\libraries -fqbn=espressif:esp32:heltec_wifi_lora_32:FlashFreq=80,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\admin\AppData\Local\Temp\arduino_build_854219 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\admin\Documents\Arduino\sketch_mar03a\sketch_mar03a.ino
Using board 'heltec_wifi_lora_32' from platform in folder: C:\Users\admin\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\admin\Documents\Arduino\hardware\espressif\esp32
Detecting libraries used...
"C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/asio" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp-tls" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_event" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_http_client" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_http_server" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_https_ota" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_https_server" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_ringbuf" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freemodbus" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/idf_test" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/libsodium" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/micro-ecc" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mqtt" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\admin\Documents\Arduino\hardw [... cut ...] 2\cores\esp32" "-IC:\Users\admin\Documents\Arduino\hardware\espressif\esp32\variants\heltec_wifi_lora_32" "C:\Users\admin\AppData\Local\Temp\arduino_build_854219\sketch\sketch_mar03a.ino.cpp" -o "C:\Users\admin\AppData\Local\Temp\arduino_build_854219\preproc\ctags_target_for_gcc_minus_e.cpp"
C:\Users\admin\Documents\Arduino\sketch_mar03a\sketch_mar03a.ino:6:41: fatal error: ulptool.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board Heltec_WIFI_LoRa_32.

i4things, 2019-03-03:
as per the instructions ulptool installed in  : 

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32\tools\ulptool

and the ulptool.h  is located in :  C:\Users\admin\Documents\Arduino\hardware\espressif\esp32\tools\ulptool\src\include\ulptool\ulptool.h

robouden, 2019-03-03:
I have similar issues with OS X. :(

i4things, 2019-03-04:
I've copy the ulptool.h in the directory where the sketch is as a temporary measure and now goes a little bit further... but the library still need fixing... the libs assumes that I have python installed : 

exec: "python": executable file not found in %PATH%
Error compiling for board Heltec_WIFI_LoRa_32.

will install python and try again (usually the esp32 libs use the local copy in the lib)

duff2013, 2019-03-04:
First you need to set your python PATH variable in windows, just search for it. 

Second, you have to install Arduino with the board manager and you directory of esp files needs to be the same as the README instructions.

Third, when testing use the ulp_README.ino example.

duff2013, 2019-03-04:
@robouden you have not provided any information so I can not help.

i4things, 2019-03-04:
Python is sorted but now I have issue with another thing - filed another ticket..

btw: using the board manager is NOT A OPTION as the max version there is 1.0.1 and it HAS CRITICAL issues like millis() problem and no timout when a wifi client is not connecting.. etc.

robouden, 2019-03-09:
duff2013,

Hereby my Mac OSX folder structure you requested. Sorry for the late reply.

[image]

duff2013, 2019-03-11:
@robouden it looks as if you don't have the esp32 core installed? Under the folder **1.0.1** all you have is the platform.txt file.