duff2013/ulptool #32: The lib requires Python to be installed on the computer

i4things, 2019-03-04:
usually the ESP32 libs have embedded python...

exec: "python": executable file not found in %PATH%
Error compiling for board Heltec_WIFI_LoRa_32.

Kei1yn, 2019-03-20:
I did install python 2.7 and still nothing.
Do you know how to fix it?

duff2013, 2019-03-21:
You need to fix your python path variable on your machine first.

leoNavarro95, 2019-05-27:
Which version of python is required? 
I have installed the 3.7 version of python and add to de PATH system variable, and still have problems when compilin the .py files:

Traceback (most recent call last):

  File "C:\Users\Leonardo\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 503, in <module>

    main(sys.argv[1:])

  File "C:\Users\Leonardo\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 102, in main

    build_ulp(PATHS, ulp_files, board_options, True)

  File "C:\Users\Leonardo\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 230, in build_ulp

    fsym.write(out)

TypeError: write() argument must be str, not bytes

exit status 1
Error compiling for board Heltec_WIFI_LoRa_32.

duff2013, 2019-05-28:
python 2.7 currently.