duff2013/ulptool #33: esp32ulp_build_recipe.py': [Errno 2] No such file or directory

i4things, 2019-03-04:
C:\Users\admin\Documents\Arduino\hardware\espressif\esp32 -u {runtime.tools.ulptool.path}/esp32ulp-elf-binutils/bin/ -x C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/ -t {runtime.tools.ulptool.path}/ --DF_CPU=240000000L --DARDUINO=10805 --DARDUINO_=Heltec_WIFI_LoRa_32 --DARDUINO_ARCH_=ESP32 --DARDUINO_BOARD="Heltec_WIFI_LoRa_32" --DARDUINO_VARIANT="heltec_wifi_lora_32"
python: can't open file '{runtime.tools.ulptool.path}/esp32ulp_build_recipe.py': [Errno 2] No such file or directory

exit status 2
Error compiling for board Heltec_WIFI_LoRa_32.

duff2013, 2019-03-04:
Unfortunately I don't support non-board manager installs. Hopefully they will come out soon with new board manager version that fixes those issues you raised about the esp32 core for Arduino.

duff2013, 2019-03-05:
Just F.Y.I you can use the board manager to download pre release that have the millis fix. I can't say that it will work but assume it will since they haven't changed anything that I can see that would break the current ulptool.

i4things, 2019-03-05:
at the moment I sorted the problem using the standard macros :  

https://github.com/i4things/NodeAPI/tree/master/examples/ESP32/ULP

Currently in the esp32 world is very active and I need to be able to choose between build/release very flexible - sometime I also need to apply fixes my self as in the Radiohead library...

duff2013, 2019-03-05:
I understand, since its open source you can hack the ulptool to use it with non standard installs I just don't want to support that since it opens up can of worms for me trouble shoot.

i4things, 2019-03-05:
mmmm I have actually followed the instruction from expressif exactly - e.g. it should be standard install : https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

duff2013, 2019-03-05:
The preferred way according the [README.md](https://github.com/espressif/arduino-esp32/blob/master/README.md) is using the Arduino Board Manager. Actually if you have both ways installed the Arduino will default to the board manager install of the esp32.