duff2013/ulptool #35: Compile error first try

Kei1yn, 2019-03-20:
I just installed your solution for ULP, but I can not compile even the readme example:
I did installed python 2.7 

Arduino log:

Arduino: 1.8.9 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default, QIO, 80MHz, 4MB (32Mb), 921600, None"

Traceback (most recent call last):

  File "C:\Users\keilyn\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 503, in <module>

    main(sys.argv[1:])

  File "C:\Users\keilyn\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 102, in main

    build_ulp(PATHS, ulp_files, board_options, True)

  File "C:\Users\keilyn\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 133, in build_ulp

    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)

  File "C:\Python27\lib\subprocess.py", line 394, in __init__

    errread, errwrite)

  File "C:\Python27\lib\subprocess.py", line 644, in _execute_child

    startupinfo)

WindowsError: [Error 2] The system cannot find the file specified

exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Does someone know what to do?
Thanks

Kei1yn, 2019-03-21:
Found out it is caused by Arduino IDF, error above shows only in arduino 1.8.9.
For 1.8.8 works just fine!

duff2013, 2019-03-21:
Huh just saw your error, thats weird I have not downloaded 1.8.9 yet but will soon.

gertlynge, 2019-06-09:
Hi, seems:

"C:\Users\<your user name>\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src\bin"

should just be copied to the catalog:

"C:\Users\<your user name>\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src\esp32ulp-elf-binutils\bin"

At least then it will compile without errors.

Best Regards.
www.dabbler.dk

nsollars, 2019-08-03:
These fixes do not work for me on Linux I get the following,

Traceback (most recent call last):
  File "/home/nsollars/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 560, in <module>
    main(sys.argv[1:])
  File "/home/nsollars/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 106, in main
    build_ulp(PATHS, ulp_files, board_options, True)
  File "/home/nsollars/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 137, in build_ulp
    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)
  File "/usr/lib64/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib64/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
exit status 1
Error compiling for board WEMOS LOLIN32.

I have edited the platform.local.txt file and what nots..

If you need more debugging I can provide, I also verified I am using 1.8.9 also

Nige

duff2013, 2019-08-03:
@nsollars,  What version of ulptool do you have?

duff2013, 2019-08-03:
Looks like the python can't find binutils, can you give me the whole path you installed binutils?

nsollars, 2019-08-04:
So here is the path to the binutils,
  /home/nsollars/.arduino15/packages/esp32/tools/ulptool/src/esp32s2ulp-elf-binutils

The versions of ulptool and bin utils are the latest listed.

duff2013, 2019-08-05:
@nsollars Can you copy the error from ***"Traceback (most recent call last):"*** back to where you find ***"-b"***. That will give me more info on the paths that are being supplied to the script.

nsollars, 2019-08-05:
@duff2013 I found the issue and its subtle, so the new binutils have been renamed ( directory and contents ), with an extra s2 like so,

esp32s2ulp-elf-binutils
esp32s2ulp-elf-addr2line

removing that from the directory name and cp'ing each file again removing that, and we are all good.  Like I said subtle ;).

Nige

duff2013, 2019-08-06:
Ah I see, thanks for following up with this!

BilalChaudry9, 2020-04-07:
I got the Same problem. I have Installed python v2.7 and added path to it.... Further I have tried some solution related to this ...
Arduino Logs:

Traceback (most recent call last):

  File "C:\Users\Abdul Rafey\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 560, in <module>

    main(sys.argv[1:])

  File "C:\Users\Abdul Rafey\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 106, in main

    build_ulp(PATHS, ulp_files, board_options, True)

  File "C:\Users\Abdul Rafey\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 127, in build_ulp

    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.STDOUT,shell=False)

  File "C:\Python27\lib\subprocess.py", line 672, in __init__

    errread, errwrite)

  File "C:\Python27\lib\subprocess.py", line 882, in _execute_child

    startupinfo)

WindowsError: [Error 2] The system cannot find the file specified

exit status 1
Error compiling for board ESP32 Dev Module.

BilalChaudry9, 2020-04-09:
Figured it out myself. There was a blank space b/w my username of machine... Renaming my User name with no Blank space resolved this issue.... 
Now that program is compiled and uploaded... My Esp32 is continuously being reset....

mahesh2000, 2021-07-27:
i have a different solution. ensure that you're picking _binutils-**esp32ulp**-xxx.zip_ and not _binutils-**esp32s2ulp**-xxx.zip_. They look almost the same.  See https://github.com/espressif/binutils-esp32ulp/releases/tag/v2.28.51-esp-20190801:
_binutils-**esp32ulp**-win32-2.28.51-esp-20190801.zip_ and not _binutils-**esp32s2ulp**-win32-2.28.51-esp-20190801.zip_. 

I still can't get the code to compile, I'm stuck at undefined reference to `ulp_entry'.

All of this worked a year ago...
.

770318926, 2022-03-08:
File "C:\Users\ASUS\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src\esp32ulp_build_recipe.py", line 305
    print e
    ^^^^^^^
I copied the files according to the manual installation steps and created a project, but the following error occurred after copying the code  

SyntaxError: Missing parentheses in call to 'print'. Did you mean print(...)?
exit status 1
Error compiling for development board ESP32 Dev Module.

770318926, 2022-03-08:
I copied the files according to the manual installation steps and created a project, but the following error occurred after copying the code.

File "C:\Users\ASUS\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src\esp32ulp_build_recipe.py", line 305
print e
^^^^^^^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(...)?
exit status 1
Error compiling for development board ESP32 Dev Module.

Lowku, 2022-05-09:
Hi, I am getting the exact same issues as mentioned above. Has anyone resolved these yet?

Thank you