duff2013/ulptool #36: fsym.write(out) TypeError argument must be str, not bytes

gerardpalk, 2019-03-31:
This looks like a great library. Thanks for the work on it.  However, I just tried to install ulptool.  On compiling ulp_README I am getting a TypeError: write() argument must be str, not bytes.  
Board is LOLIN D32 PRO.  ulptool v2.3.0, Board Manager ESP32, Version 1.0.1, Arduino 1.8.9.

Any help would be appreciated. I am very new to the ESP32.

`Arduino: 1.8.9 (Windows 10), Board: "LOLIN D32 PRO, Disabled, Default, 80MHz, 921600, None"

`Traceback (most recent call last):

  File "C:\Users\gerard\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 503, in <module>

    main(sys.argv[1:])

  File "C:\Users\gerard\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 102, in main

    build_ulp(PATHS, ulp_files, board_options, True)

  File "C:\Users\gerard\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 230, in build_ulp

    fsym.write(out)

TypeError: write() argument must be str, not bytes

exit status 1
Error compiling for board LOLIN D32 PRO.`

duff2013, 2019-04-01:
I have never seen this error before that I remember, what version of python do you have?

gerardpalk, 2019-04-01:
I am running Python 3.6.1.

duff2013, 2019-04-01:
This was made for python 2.7, can you use that? I will upgrade to python 3 sometime soon since 2 is not going to be supported anymore soon but when I can't tell you now.

gerardpalk, 2019-04-01:
I am sure I can install Python 2.7.  Thanks for the advice.

gerardpalk, 2019-04-02:
Thanks again for your help.  Installed Python 2.7 and all is good.

closed 2019-04-02 by gerardpalk

tilman1, 2021-02-07:
I received the same error message, and I am running Python 2.7.18 on ubuntu 20.4. The board selected is DOIT DEVKIT V1.0:
Arduino: 1.8.13 (Linux), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

I fixed it by adding one 'b'  to esp32ulp_build_recipe.py:line 233:
line 233: with open(file_names_constant['sym'],"w**b**") as fsym:

In addition I had to add brackets to 2 print commands:
line 305: print **(**e**)**
line 316: print **(**cmd[0]**)**

With this, it compiles. Whether the object code works, i.e. runs on the DevKit, I am yet to try.

[esp32ulp_build_recipe.py.gz](https://github.com/duff2013/ulptool/files/5940002/esp32ulp_build_recipe.py.gz)