duff2013/ulptool #38: Error with python using Adafruit ESP32 Feather Board

psavidar, 2019-04-25:
Error Message: Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None"

Traceback (most recent call last):

  File "C:\Users\Daniel Davidar\Documents\ArduinoData\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 503, in <module>

    main(sys.argv[1:])

  File "C:\Users\Daniel Davidar\Documents\ArduinoData\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 102, in main

    build_ulp(PATHS, ulp_files, board_options, True)

  File "C:\Users\Daniel Davidar\Documents\ArduinoData\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 123, in build_ulp

    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.STDOUT,shell=False)

  File "C:\Users\Daniel Davidar\AppData\Local\Programs\Python\Python37-32\lib\subprocess.py", line 775, in __init__

    restore_signals, start_new_session)

  File "C:\Users\Daniel Davidar\AppData\Local\Programs\Python\Python37-32\lib\subprocess.py", line 1178, in _execute_child

    startupinfo)

FileNotFoundError: [WinError 2] The system cannot find the file specified

exit status 1
Error compiling for board Adafruit ESP32 Feather.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

duff2013, 2019-04-26:
did you set python27 environment variable for windows?

psavidar, 2019-04-26:
Not sure where to do that. DO you have any tips?

psavidar, 2019-04-26:
I see you have answered a question very similar: https://github.com/duff2013/ulptool/issues/6
We both have nearly identical problems, but the file path for this version seems slightly different.
Do you know what I am doing incorrectly?

duff2013, 2019-04-26:
https://www.pythoncentral.io/add-python-to-path-python-is-not-recognized-as-an-internal-or-external-command/

What esp core version do you have?
Make sure you install esp core from the Arduino Library manager.

psavidar, 2019-04-26:
Ok, I used the Arduino Board Manager. I have version 1.0.2.

psavidar, 2019-04-26:
I set the path earlier.

duff2013, 2019-04-27:
ok, good what path did you install the ULPTOOL?

psavidar, 2019-04-27:
C:\Users\Daniel Davidar\AppData\Local\Arduino15\packages\esp32\tools\ulptool .

duff2013, 2019-04-28:
thats the right path but the space between Daniel and Davidar might be an issue? You also put platform.local.txt in:
C:\Users\Daniel Davidar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\

borstan, 2019-04-29:
I did everything as mentioned and the only thing I am suspicious about is that in my system folder name is used letter **Č** that might not be recognizable. Same with space that Daniel Davidar has there. I try to solve it and let you know.

borstan, 2019-05-02:
I renamed the folder and the error persists.

Traceback (most recent call last):

  File "C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 503, in <module>

    main(sys.argv[1:])

  File "C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 102, in main

    build_ulp(PATHS, ulp_files, board_options, True)

  File "C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 230, in build_ulp

    fsym.write(out)

TypeError: write() argument must be str, not bytes

exit status 1
Nastala chyba při kompilaci u desky SparkFun ESP32 Thing.

borstan, 2019-05-02:
UPDATE: When I reinstalled the version of my Python to 2.7 everything works fine! My core version is 1.0.2.

ajvuik, 2019-05-12:
Hello, I had this exact same error, but on my part it was because my %appdata% directory has spaces in it and Python seems to be having problems with spaces in paths. 
Making the Arduino IDE portable solved my problem, because then I can change the packages directory.
I am using Window 10

duff2013, 2019-05-12:
Good to hear @borstan, @ajvuik I'll work on the 'space' issue'.

psavidar, 2019-05-21:
Thank you very much for your help. It is running and compiling the ULP. I re-installed ESP 1.0.1 and used version 2.1.0 for the ulptool. I then created a new account without the spaces. This must have been the error, very thankful for your help.

closed 2019-05-21 by psavidar

duff2013, 2019-05-21:
Oh no problem glad it worked, thanks for following up with me!