duff2013/ulptool #40: region `ram' overflowed

borstan, 2019-05-02:
Hello guys, I resolved the recent problem with python but now I am getting an error regarding the memory overflow. I can run ulp_README.ino with no problem but with examples like ulp_adc.ino or ulp_I2C.ino I am getting this error. Can anyone help me with this issue? Thank you for any advice.

PS: I read [#18](https://github.com/duff2013/ulptool/issues/18) but I don't understand the problem resolving there.

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Boris\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Boris\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Boris\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32thing:FlashFreq=80,PartitionScheme=default,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\Boris\AppData\Local\Temp\arduino_build_490723 -warnings=none -build-cache C:\Users\Boris\AppData\Local\Temp\arduino_cache_979679 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Boris\Documents\Arduino\ulp_examples\ulp_adc\ulp_adc.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Boris\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Boris\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Boris\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32thing:FlashFreq=80,PartitionScheme=default,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\Boris\AppData\Local\Temp\arduino_build_490723 -warnings=none -build-cache C:\Users\Boris\AppData\Local\Temp\arduino_cache_979679 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Boris\Documents\Arduino\ulp_examples\ulp_adc\ulp_adc.ino
Using board 'esp32thing' from platform in folder: C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Using core 'esp32' from platform in folder: C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Detecting libraries used...
"C:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/config" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_trace" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_update" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/asio" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bootloader_support" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bt" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/coap" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/console" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/driver" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp-tls" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp32" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_event" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_http_client" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_http_server" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_https_ota" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/ethernet" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/expat" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/fatfs" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/freemodbus" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/freertos" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/heap" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/idf_test" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/jsmn" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/json" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/libsodium" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/log" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/lwip" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mbedtls" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mdns" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/micro-ecc" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mqtt" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/newlib" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/nghttp" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/nvs_flash" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/openssl" "-IC:\\Users\\Boris\\AppData\\Local\\Arduino15\\package [... cut ...]
C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp-elf-binutils/bin/esp32ulp-elf-ld -o ulp_main.elf -A elf32-esp32ulp -Map=ulp_main.map -T ulp_main.common.ld adc.ulp.o
C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp-elf-binutils/bin/esp32ulp-elf-ld: ulp_main.elf section `.data' will not fit in region `ram'

C:\Users\Boris\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp-elf-binutils/bin/esp32ulp-elf-ld: region `ram' overflowed by 88 bytes

duff2013, 2019-05-03:
try increasing the ulp ram by finding CONFIG_ULP_COPROC_RESERVE_MEM in sdkconfig.h

borstan, 2019-05-07:
Thank you, now it works!

closed 2019-05-07 by borstan