duff2013/ulptool #42: Having no luck installing ulptool

vangalvin, 2019-08-03:
Just to be sure I did not have anyting weird on my main machine I created a new virtual machine in Hyper-V. Did a fresh install using the following...

Pyton 2.7
Arduino IDE 1.8.9
ESP32 installed from the board manager

I keep getting the following, I have tried to step thoguht the install instructions be get teh same result no matter what I try.

`Traceback (most recent call last):
  File "C:\Users\Andrew Hooper\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 560, in <module>
    main(sys.argv[1:])
  File "C:\Users\Andrew Hooper\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 106, in main
    build_ulp(PATHS, ulp_files, board_options, True)
  File "C:\Users\Andrew Hooper\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 127, in build_ulp
    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.STDOUT,shell=False)
  File "C:\Python27\lib\subprocess.py", line 394, in __init__
    errread, errwrite)
  File "C:\Python27\lib\subprocess.py", line 644, in _execute_child
    startupinfo)
WindowsError: [Error 2] The system cannot find the file specified
exit status 1
Error compiling for board ESP32 Dev Module.
`
I read most of the other instructions and faults that others have had however none of the fixes seemed to work.

duff2013, 2019-08-03:
What version of ulptool do you have?

vangalvin, 2019-08-03:
The ZIP file says its 2.4.0 however in the platform.local.txt it says 2.3.0

Last night I deleted everyting then then tried to install the full espressif tool chain, I must have done something wring as now i have issues with the path and finding "ulptool.h"

I have a feeling you may have to write a book. "The dark arts of setting up for ULP programming" :)

vangalvin, 2019-08-04:
OK, so days of messing about trying to get it working and thins morning following the same proceedure as I did thinking i would try and hard code the paths and sudenly it works lol...
The only diference was that I dowloaded a fresh copy of esp32ulp-elf-binutils. I really have no idea what I did that was diferent this time.

However thank you, And with luck this will allow me to read the temperatures from the sensors without having to wake up the main processor :)

I still have to work out if I can store the sampled temperatures either in the SPIFFS or flash while its in sleep mode, from what I read the ULP has access to it however have not been able to find an example of this meing done as yet. 

My unit will be opperating in -30degrees C and hoping to also sense the temperature inside the case, if its below a specific point hoping to be able to turn on an internal heater and heat the system before the main processor kicks in as there have been a few instances of chips cracking due to thermal shock.

duff2013, 2019-08-04:
Is there a space between "Andrew" and "Hooper" in your path?

vangalvin, 2019-08-04:
No, I did spot that issue and checked to make sure.
Same with the other machine as I made sure to install everyting on paths to make sure that there were no spaces in the file names. 

It seems to be working now. I wish I knew what I did diferently the last time.

f-e-l-i-x, 2019-11-09:
I had the same error, because I accidentally downloaded binutils-esp32s2ulp-win32-2.28.51-esp-20190801.zip and not binutils-esp32ulp-win32-2.28.51-esp-20190801.zip

They are on the same download page and the additional "s2" in the filename is difficult to spot for some ...

Using the correct version fixed it.

fe7565, 2019-11-24:
I had the same problem with the files, took me half a day to figure it out.  Do not know what genius named two files with over 50 characters almost the same name with the exception of the "2".