duff2013/ulptool #44: Arduino IDE from MS Store

hosek, 2019-08-06:
Just quick note about issue I have found yesterday while helping friend to have ulptool up and running.

Arduino IDE from MS Store is using different paths to save ESP32 core 
`C:\Users\<USERNAME>\Documents\ArduinoData`
instead of 
`C:\Users\<USERNAME>\AppData ... `

Its tricky to find and maybe it could added to readme, that users should use only the version from official website

duff2013, 2019-08-06:
Ok I do say in the readme: ***"This guide assumes you installed the esp32 core with the preferred method of the board manager."*** but I guess it could be more explicitly stated in the future, thanks for bringing me this issue.

hosek, 2019-08-06:
Problem is that esp32 core was installed via boards manager, but into different location - MS Store Arduino IDE is probably some custom build of Arduino IDE with different default paths.

duff2013, 2019-08-06:
Oh I see, I'll take that into account on the next update in a week or two.