duff2013/ulptool #45: Can't compile ULPCC code

nzagorec, 2019-08-18:
Hi @duff2013 

I compile ulp examples normally but am unable to compile ulpcc. I keep getting the following error:

```
/tmp/arduino_build_68295/sketch/ulpcc_counter.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `ulp_entry'
/tmp/arduino_build_68295/sketch/ulpcc_counter.ino.cpp.o:(.literal._Z4loopv+0x0): undefined reference to `ulp_counter'
```

It seems that code from ulp_counter.c never gets loaded. If I manually add #define _ULPCC_ on top of the file I get this:

`ulp_counter.c:10:19: error: ulp_c.h: No such file or directory`

Are you sure you've uploaded the correct platform.local.txt file? On the first line it says version=2.3.0 but correct version is 2.4.0.

I'm using latest esp32-arudino 1.0.2 by the way.

Thanks!

duff2013, 2019-08-19:
What os are you using?

nzagorec, 2019-08-19:
I'm using Linux.

duff2013, 2019-08-19:
Can you turn verbose compiling for the Arduino IDE and paste the full output.

nzagorec, 2019-08-19:
Here it is...

```
/home/user/arduino-1.8.8/arduino-builder -dump-prefs -logger=machine -hardware /home/user/arduino-1.8.8/hardware -hardware /home/user/.arduino15/packages -tools /home/user/arduino-1.8.8/tools-builder -tools /home/user/arduino-1.8.8/hardware/tools/avr -tools /home/user/.arduino15/packages -built-in-libraries /home/user/arduino-1.8.8/libraries -libraries /home/user/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10808 -build-path /tmp/arduino_build_268321 -warnings=none -build-cache /tmp/arduino_cache_113015 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/user/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/home/user/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=/home/user/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/home/user/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.mkspiffs.path=/home/user/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/user/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /home/user/Arduino/ulp_examples/ulpcc/ulpcc_counter/ulpcc_counter.ino
/home/user/arduino-1.8.8/arduino-builder -compile -logger=machine -hardware /home/user/arduino-1.8.8/hardware -hardware /home/user/.arduino15/packages -tools /home/user/arduino-1.8.8/tools-builder -tools /home/user/arduino-1.8.8/hardware/tools/avr -tools /home/user/.arduino15/packages -built-in-libraries /home/user/arduino-1.8.8/libraries -libraries /home/user/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10808 -build-path /tmp/arduino_build_268321 -warnings=none -build-cache /tmp/arduino_cache_113015 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/user/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/home/user/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=/home/user/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/home/user/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.mkspiffs.path=/home/user/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/user/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /home/user/Arduino/ulp_examples/ulpcc/ulpcc_counter/ulpcc_counter.ino
Using board 'esp32' from platform in folder: /home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2
Using core 'esp32' from platform in folder: /home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2
Detecting libraries used...
/home/user/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/config -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/app_trace -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/app_update -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/asio -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/bootloader_support -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/bt -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/coap -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/console -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/driver -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/esp-tls -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/esp32 -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/esp_adc_cal -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/esp_event -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/esp_http_client -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/esp_http_server -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/esp_https_ota -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/esp_ringbuf -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/ethernet -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/expat -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/fatfs -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/freemodbus -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/freertos -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/heap -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/tools/sdk/include/idf_test -I/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.2/t [... cut ...] dtls -llwip -lnet80211 -lesp_http_server -lpthread -ljson -lesp_https_ota -lstdc++ -Wl,--end-group -Wl,-EL -o /tmp/arduino_build_268321/ulpcc_counter.ino.elf
/tmp/arduino_build_268321/sketch/ulpcc_counter.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `ulp_entry'
/tmp/arduino_build_268321/sketch/ulpcc_counter.ino.cpp.o:(.literal._Z4loopv+0x0): undefined reference to `ulp_counter'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.
```

duff2013, 2019-08-19:
The script is not seeing the .c ulp file, can you post that ulp_counter.c file as you have it now.

nzagorec, 2019-08-20:
I haven't made any changes to it, I'm using the script from ulp_examples folder.

```
/*
 * do not add code above here all 
 * ulp c code must be between this 
 * ifdef.
 */
#ifdef _ULPCC_
// must include ulpcc helper functions
#include <ulp_c.h>

// global variable that the main processor can see
unsigned counter = 0;

// all ulpcc programs have have this function
void entry() {
  // increment counter
  counter++;
}
#endif // do not add code after here
```

duff2013, 2019-08-20:
Ok, it maybe something stupid I did, remove the comment before "#ifdef _ULPCC_" and try again.

duff2013, 2019-08-20:
The script looks for the ifdef on the first line and endif on the last line. I'll update my examples soon. Hope this fixes your problem, let me know.

nzagorec, 2019-08-20:
You mean to remove this:

```
/*
 * do not add code above here all 
 * ulp c code must be between this 
 * ifdef.
 */
```

Tried, same thing.
If I add `#define _ULPCC_` on top (or remove ifdef/endif completely), I get this error:

```
ulp_counter.c:10:19: error: ulp_c.h: No such file or directory
compilation terminated.
exit status 1
ulp_c.h: No such file or directory
```

duff2013, 2019-08-20:
yes remove that comment at the top
```
/*
 * do not add code above here all 
 * ulp c code must be between this 
 * ifdef.
 */
```

and don't add the define _ULPCC_

nzagorec, 2019-08-20:
Read my last comment again :smiley: 

First I tried removing the comment without adding anything - it still gave me the same error as before (`undefined reference to `ulp_entry'`).

Then I added a define and got new error (`ulp_c.h: No such file or directory`).

duff2013, 2019-08-20:
I did read but you are doing thing I did not say to do

duff2013, 2019-08-20:
The compiler supplies that define not the **USER!**

nzagorec, 2019-08-20:
@duff2013, answer to your question (to remove the comment) is: I tried and it made no difference, I still got the same error as with the comment there (`undefined reference to ulp_entry'``).

After I tried removing the comment, I put the comment back and added a `#define`.

The reason why I added that `#define` is to show you that the compiler obvously did **NOT** supply it because when the **USER** supplies it, the error is different.

This implies that the compiler never passes this line: `#ifdef _ULPCC_`
Because _ULPCC_ is **NOT** defined.

duff2013, 2019-08-20:
The Arduino output you supplied says that no ulp c files where found, thats python not the compiler checking for those files, this is the actual error and is exactly what I saw when I just ran the counter example with the comment above the ifdef. 

The compiler which I compiled does supply the ```_ULPCC__``` define, though in this case you are not even getting to that point. If you look at the Arduino output that you supplied you will see no call to lcc compiler, that tells me that the script did not find any ulp c files.

nzagorec, 2019-08-20:
The problem is not only in comment above `#ifdef _ULPCC_` but also in comment after `#endif`.

I think you should re-do this part:
```
                top = f.readline().strip()
                bottom = f.readlines()[-1].strip()
                if top == "#ifdef _ULPCC_":
                    if bottom == "#endif":
                        ulpcc_files.append(file)
```

duff2013, 2019-08-21:
Yes that is true, did you get it to work yet?

nzagorec, 2019-08-21:
Yes, I removed both comments and ulpcc_counter works perfectly.