duff2013/ulptool #46: Unable to compile ULP project

keithr0, 2019-10-07:
Trying to compile several ULP examples I get this error  (Windows 10 32 bit Python 3.7-32):-

`Arduino: 1.8.10 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

D:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Program Files\Arduino\hardware -hardware C:\Users\Keith\AppData\Local\Arduino15\packages -tools D:\Program Files\Arduino\tools-builder -tools D:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\Keith\AppData\Local\Arduino15\packages -built-in-libraries D:\Program Files\Arduino\libraries -libraries C:\Users\Keith\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10810 -build-path C:\Users\Keith\AppData\Local\Temp\arduino_build_387794 -warnings=none -build-cache C:\Users\Keith\AppData\Local\Temp\arduino_cache_124870 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\Keith\Documents\Arduino\ulp_test\ulp_test.ino
D:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware D:\Program Files\Arduino\hardware -hardware C:\Users\Keith\AppData\Local\Arduino15\packages -tools D:\Program Files\Arduino\tools-builder -tools D:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\Keith\AppData\Local\Arduino15\packages -built-in-libraries D:\Program Files\Arduino\libraries -libraries C:\Users\Keith\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10810 -build-path C:\Users\Keith\AppData\Local\Temp\arduino_build_387794 -warnings=none -build-cache C:\Users\Keith\AppData\Local\Temp\arduino_cache_124870 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\Keith\Documents\Arduino\ulp_test\ulp_test.ino
Using board 'esp32' from platform in folder: C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Detecting libraries used...
"C:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nghttp" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nvs_flash" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/openssl" "-IC:\\Users\\Keith\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protobuf-c" "-IC:\\Users\\Keith\\AppData\\Local\\A [... cut ...]
  File "C:\Users\Keith\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 137, in build_ulp

    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)

  File "D:\Program Files\Python\Python37-32\lib\subprocess.py", line 775, in __init__

    restore_signals, start_new_session)

  File "D:\Program Files\Python\Python37-32\lib\subprocess.py", line 1178, in _execute_child

    startupinfo)

FileNotFoundError: [WinError 2] The system cannot find the file specified

exit status 1
Error compiling for board ESP32 Dev Module.`

duff2013, 2019-10-07:
It looks like you are using python37?

keithr0, 2019-10-08:
That is true.

keithr0, 2019-10-08:
Exactly the same with 2.7. I don’t know if it being 32 bit makes any difference but I wouldn’t have thought so.

duff2013, 2019-10-10:
can you give me the exact name of the binutils folder, is it "esp32s2ulp-elf-binutils"?

keithr0, 2019-10-11:
Yes, it is d:\esp32s2ulp-elf-binutils.

keithr0, 2019-10-11:
Correction:
C:\users\keith\AppData\Local\Arduino15\packages\esp32\hardware\esp32\esp32s2ulp-elf-binutils

duff2013, 2019-10-11:
Can you try downloading binutils that does not have s2 in the name from here and let me know.

[https://github.com/espressif/binutils-esp32ulp/releases/tag/v2.28.51-esp-20190801](https://github.com/espressif/binutils-esp32ulp/releases/tag/v2.28.51-esp-20190801)

keithr0, 2019-10-12:
Unfortunately, the same result.

keithr0, 2019-10-12:
cmds at line 115 value is:
{‘ULPMAPGEN’: ‘C:\users\keith\appdata\local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\esp32_mapgen.py’}
esp32_mapgen.py is located at:
C:\users\keith\appdata\local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\ulptool\src\esp32_mapgen.py

I don’t know if this is related.

keithr0, 2019-10-13:
I tried copying the file to the other place but no change. I will be without internet for the next few days, I’ll continue to fault find.

duff2013, 2019-10-14:
Ok that sucks but you need the non s2 version of binutils for sure and it needs to be in the directory ...\tools\ulptool\src\

your "esp32_mapgen.py" needs to be in the same directory as the binutils no need to move it when you install it is in the correct place.

d0ct0rd-dalimo, 2019-10-15:
I've modified the esp32ulp_build_recipe.py in order to show path it use, so it uses wrong path:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages -hardware F:\Kuzmichev\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries F:\Kuzmichev\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=noota_3g,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10810 -build-path C:\Users\KUZMIC~1\AppData\Local\Temp\arduino_build_268611 -warnings=none -build-cache C:\Users\KUZMIC~1\AppData\Local\Temp\arduino_cache_607990 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose F:\Kuzmichev\Arduino\ulp_examples\ulp_adc\ulp_adc.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages -hardware F:\Kuzmichev\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries F:\Kuzmichev\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=noota_3g,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10810 -build-path C:\Users\KUZMIC~1\AppData\Local\Temp\arduino_build_268611 -warnings=none -build-cache C:\Users\KUZMIC~1\AppData\Local\Temp\arduino_cache_607990 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose F:\Kuzmichev\Arduino\ulp_examples\ulp_adc\ulp_adc.ino
Using board 'esp32' from platform in folder: C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Detecting libraries used...
"C:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\KuzmichevDO [... cut ...]
    build_ulp(bpath, ppath, xpath, upath, ulp_files, board_options)

  File "C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 118, in build_ulp

    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)

  File "C:\Python27\lib\subprocess.py", line 390, in __init__

    errread, errwrite)

  File "C:\Python27\lib\subprocess.py", line 640, in _execute_child

    startupinfo)

WindowsError: [Error 2] 

exit status 1
Ошибка компиляции для платы ESP32 Dev Module.

d0ct0rd-dalimo, 2019-10-15:
Maybe, tilda breaks the script down: 

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages -hardware F:\Kuzmichev\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries F:\Kuzmichev\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=noota_3g,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10810 -build-path C:\Users\KUZMIC~1\AppData\Local\Temp\arduino_build_268611 -warnings=none -build-cache C:\Users\KUZMIC~1\AppData\Local\Temp\arduino_cache_607990 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose F:\Kuzmichev\Arduino\ulp_examples\ulp_adc\ulp_adc.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages -hardware F:\Kuzmichev\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries F:\Kuzmichev\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=noota_3g,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10810 -build-path C:\Users\KUZMIC~1\AppData\Local\Temp\arduino_build_268611 -warnings=none -build-cache C:\Users\KUZMIC~1\AppData\Local\Temp\arduino_cache_607990 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose F:\Kuzmichev\Arduino\ulp_examples\ulp_adc\ulp_adc.ino
Using board 'esp32' from platform in folder: C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Using core 'esp32' from platform in folder: C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4
Detecting libraries used...
"C:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\KuzmichevDO\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\es [... cut ...]
    build_ulp(bpath, ppath, xpath, upath, ulp_files, board_options)

  File "C:\Users\KuzmichevDO\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 118, in build_ulp

    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)

  File "C:\Python27\lib\subprocess.py", line 390, in __init__

    errread, errwrite)

  File "C:\Python27\lib\subprocess.py", line 640, in _execute_child

    startupinfo)

WindowsError: [Error 2] 

exit status 1
Ошибка компиляции для платы ESP32 Dev Module.

keithr0, 2019-10-20:
Thanks for the help, the first problem was the 2 in the name, the second was my finger trouble. I can compile now thanks

closed 2019-10-20 by keithr0

swaraj429, 2020-05-07:
how to change python 3.8 to 2.7 
i have both installed??

peteDDD, 2023-08-26:
In C:\Users\USERNAME\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src\esp32ulp_mapgen.py

change:
#!/usr/bin/env python

to

#!/usr/bin/env python2.7

(assuming you also have phthon 2.7 installed)