duff2013/ulptool #47: Compile Errors

mgamache, 2019-11-09:
I am Using IDE 1.8.10 and Python 3.8.0 and esp32 Board 1.0.2. I fixed a few Print statement errors, but I am still getting errors . I may have something miss-configured or there are version issues. Any help would be appreciated... 

`Traceback (most recent call last):

  File "C:\Users\<user>\AppData\Local\arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 560, in <module>

    main(sys.argv[1:])

  File "C:\Users\<user>\AppData\Local\arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 106, in main

    build_ulp(PATHS, ulp_files, board_options, True)

  File "C:\Users\<user>\AppData\Local\arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 137, in build_ulp

    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)

  File "C:\Users\<user>\AppData\Local\Programs\Python\Python38\lib\subprocess.py", line 854, in __init__

    self._execute_child(args, executable, preexec_fn, close_fds,

  File "C:\Users\<user>\AppData\Local\Programs\Python\Python38\lib\subprocess.py", line 1307, in _execute_child

    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,

FileNotFoundError: [WinError 2] The system cannot find the file specified

exit status 1
Error compiling for board Adafruit ESP32 Feather.`

duff2013, 2019-11-13:
This error is it can't find a file, whats the name of the binutils you downloaded?

usmanshahid001, 2019-12-19:
i am facing exactly the same issue

duff2013, 2019-12-19:
see my comment and report back?

usmanshahid001, 2019-12-24:
esp32ulp-elf-binutils.downloaded the latest version.may be issue of my python version

usmanshahid001, 2019-12-27:
any help please

duff2013, 2019-12-27:
Python 3 is not really supported at this time, could be that, can you use python 2.7?

usmanshahid001, 2020-01-07:
ok i will give a try

usmanshahid001, 2020-01-10:
yes it worked with python 2.7

duff2013, 2020-01-10:
Good to hear, I'm going to close this issue now.

closed 2020-01-10 by duff2013

BilalChaudry9, 2020-04-07:
Traceback (most recent call last):

  File "C:\Users\Abdul Rafey\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 560, in <module>

    main(sys.argv[1:])

  File "C:\Users\Abdul Rafey\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 106, in main

    build_ulp(PATHS, ulp_files, board_options, True)

  File "C:\Users\Abdul Rafey\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 127, in build_ulp

    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.STDOUT,shell=False)

  File "C:\Python27\lib\subprocess.py", line 672, in __init__

    errread, errwrite)

  File "C:\Python27\lib\subprocess.py", line 882, in _execute_child

    startupinfo)

WindowsError: [Error 2] The system cannot find the file specified

exit status 1
Error compiling for board ESP32 Dev Module.

I am facing this same issue... And I have been using Python V2.7...