duff2013/ulptool #48: Compile Errors

zoran1987, 2019-12-18:
I am Using IDE 1.8.10 and Python 2.7.17 and esp32 Board 1.0.2. I can run other ulp_examples, but ulpcc "ulpcc_adc" and "ulpcc_counter" I can't compile, Can you help me?

sketch\ulpcc_adc.ino.cpp.o:(.literal._Z5setupv+0x20): undefined reference to `ulp_threshold'

sketch\ulpcc_adc.ino.cpp.o:(.literal._Z5setupv+0x24): undefined reference to `ulp_entry'

sketch\ulpcc_adc.ino.cpp.o:(.literal._Z4loopv+0x0): undefined reference to `ulp_mutex'

sketch\ulpcc_adc.ino.cpp.o:(.literal._Z4loopv+0x4): undefined reference to `ulp_adc_avg_result'

sketch\ulpcc_adc.ino.cpp.o:(.literal._Z4loopv+0x8): undefined reference to `ulp_sample_counter'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ESP32 Dev Module.

duff2013, 2019-12-19:
What OS are you using?

zoran1987, 2019-12-19:
Windows 10

zoran1987, 2019-12-19:
I think that I have the same issue as @nzagorec (Can't compile ULPCC code #45) but on windows. :)
If I remove comment after #endif and before #ifdef in upl_counter.c I got this:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\pc\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\pc\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\pc\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10810 -build-path C:\Users\pc\AppData\Local\Temp\arduino_build_933497 -warnings=none -build-cache C:\Users\pc\AppData\Local\Temp\arduino_cache_391907 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose E:\Download\ulptool-2.4.1 (2)\ulptool\src\ulp_examples\ulpcc\ulpcc_counter\ulpcc_counter.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\pc\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\pc\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\pc\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10810 -build-path C:\Users\pc\AppData\Local\Temp\arduino_build_933497 -warnings=none -build-cache C:\Users\pc\AppData\Local\Temp\arduino_cache_391907 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose E:\Download\ulptool-2.4.1 (2)\ulptool\src\ulp_examples\ulpcc\ulpcc_counter\ulpcc_counter.ino
Using board 'esp32' from platform in folder: C:\Users\pc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Using core 'esp32' from platform in folder: C:\Users\pc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Detecting libraries used...
"C:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/config" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_trace" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_update" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/asio" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bootloader_support" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bt" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/coap" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/console" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/driver" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp-tls" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp32" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/inclu [... cut ...] end-group -Wl,-EL -o "C:\\Users\\pc\\AppData\\Local\\Temp\\arduino_build_933497/ulpcc_counter.ino.elf"
C:\Users\pc\AppData\Local\Temp\arduino_build_933497\sketch\ulpcc_counter.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `ulp_entry'

C:\Users\pc\AppData\Local\Temp\arduino_build_933497\sketch\ulpcc_counter.ino.cpp.o:(.literal._Z4loopv+0x0): undefined reference to `ulp_counter'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ESP32 Dev Module.

zoran1987, 2019-12-19:
If I remove all comments from ulp_counter.c I got this:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\pc\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\pc\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\pc\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10810 -build-path C:\Users\pc\AppData\Local\Temp\arduino_build_933497 -warnings=none -build-cache C:\Users\pc\AppData\Local\Temp\arduino_cache_391907 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose E:\Download\ulptool-2.4.1 (2)\ulptool\src\ulp_examples\ulpcc\ulpcc_counter\ulpcc_counter.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\pc\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\pc\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\pc\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10810 -build-path C:\Users\pc\AppData\Local\Temp\arduino_build_933497 -warnings=none -build-cache C:\Users\pc\AppData\Local\Temp\arduino_cache_391907 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose E:\Download\ulptool-2.4.1 (2)\ulptool\src\ulp_examples\ulpcc\ulpcc_counter\ulpcc_counter.ino
Using board 'esp32' from platform in folder: C:\Users\pc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Using core 'esp32' from platform in folder: C:\Users\pc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Detecting libraries used...
"C:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/config" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_trace" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_update" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/asio" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bootloader_support" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bt" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/coap" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/console" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/driver" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp-tls" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp32" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_event" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_h [... cut ...] sp32\\tools\\ulptool\\src/esp32ulp-elf-binutils/bin/" -x "C:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/" -t "C:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\ulptool\\src/" --DF_CPU=240000000L --DARDUINO=10810 --DARDUINO_=ESP32_DEV --DARDUINO_ARCH_=ESP32 "--DARDUINO_BOARD=\"ESP32_DEV\"" "--DARDUINO_VARIANT=\"esp32\""
ulpcc is not supported on Windows

exit status 1
Error compiling for board ESP32 Dev Module.

duff2013, 2019-12-19:
Sorry but the windows ulp c compiler is not included with ulptool I only made a binary for linux and Mac. I may try to include it at a later date but now I'm too busy with other projects.

zoran1987, 2019-12-19:
Tnx for your answar, I will try with linux

closed 2019-12-19 by zoran1987