duff2013/ulptool #49: Can't compile ulp_adc- `ram' overflowed by 88 bytes

risejohan, 2019-12-21:
Hi, I have a problem that you can help me.
I am currently running the ulptools with windows 10, arduino ide v1.8.5, I modified the sdkconfig.h file in the coprocessor's memory in 2048 using version 1.0.2 and I am using the esp32ulp-elf-binutils of the s2, although I have the dual esp32, remove all the s2 as in previous forums, using python 2.7 I modified the platform.local.txt using the python compilation recommendation

compiler.s.cmd.windows = py -2 "{tools.ulptool.path} {tools.ulptool.cmd}"

but I still have the error by  region `ram' overflowed by 88 bytes

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\johan\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\johan\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\johan\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10805 -build-path C:\Users\johan\AppData\Local\Temp\arduino_build_445808 -warnings=none -build-cache C:\Users\johan\AppData\Local\Temp\arduino_cache_990957 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src\ulp_examples\ulp_adc\ulp_adc.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\johan\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\johan\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\johan\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10805 -build-path C:\Users\johan\AppData\Local\Temp\arduino_build_445808 -warnings=none -build-cache C:\Users\johan\AppData\Local\Temp\arduino_cache_990957 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src\ulp_examples\ulp_adc\ulp_adc.ino
Using board 'esp32' from platform in folder: C:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Using core 'esp32' from platform in folder: C:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Detecting libraries used...
"C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/config" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/app_trace" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/app_update" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/asio" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/bootloader_support" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/bt" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/coap" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/console" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/driver" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/esp-tls" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/esp32" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/esp_adc_cal" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/esp_event" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/esp_http_client" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/esp_http_server" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/esp_https_ota" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/esp_ringbuf" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/ethernet" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/expat" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/fatfs" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/freemodbus" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/freertos" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/heap" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/idf_test" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/jsmn" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/json" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/libsodium" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/log" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/lwip" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/mbedtls" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/mdns" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/micro-ecc" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/mqtt" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/newlib" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/nghttp" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/nvs_flash" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/openssl" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/protobuf-c" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/protocomm" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/pthread" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/sdmmc" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/smartconfig_ack" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/soc" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/spi_flash" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/spiffs" "-IC:\Users\johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2/tools/sdk/inc [... cut ...]
C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp-elf-binutils/bin/esp32ulp-elf-ld -o ulp_main.elf -A elf32-esp32ulp -Map=ulp_main.map -T ulp_main.common.ld adc.ulp.o
C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp-elf-binutils/bin/esp32ulp-elf-ld: ulp_main.elf section `.data' will not fit in region `ram'

C:\Users\johan\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp-elf-binutils/bin/esp32ulp-elf-ld: region `ram' overflowed by 88 bytes

exit status 1
Error compilando para la tarjeta ESP32 Dev Module.

**thanks to any solution**

duff2013, 2019-12-21:
Can you try to increase the sdkconfig.h memory more?

risejohan, 2019-12-27:
I already made the memory increase in the file sdkconfig.h to 2048, as it says in the intro of the program
[image]

duff2013, 2019-12-27:
Even more like 4096

risejohan, 2019-12-28:
I did it. but without any result.

duff2013, 2020-01-04:
it needs to be sdkconfig.h file