duff2013/ulptool #50: ULPCC example in README.md broken

drandreas, 2020-01-08:
The README.md contains the following snippet:

```
/*
 * do not add code above here all
 * ulp c code must be between this
 * ifdef.
 */
#ifdef _ULPCC_
// must include ulpcc helper functions
#include <ulp_c.h>

// global variable that the main processor can see
unsigned counter = 0;

// all ulpcc programs have have this function
void entry() {
  // increment counter
  counter++;
}
#endif // do not add code after here
```

However the gen_assembly expects a clean file:
```
        for file in c_files:
            with open(file, "rb") as f:
                top = f.readline().strip()
                bottom = f.readlines()[-1].strip()
                if top == "#ifdef _ULPCC_":
                    if bottom == "#endif":
                        ulpcc_files.append(file)
```

Therefore the documentation disables/breaks the compilation of `ulp_counter.c`.

Everything else works flawlessly on MacOS Catalina 👍 .

duff2013, 2020-01-08:
Thanks! just fixed it.

drandreas, 2020-01-09:
Thank you for the prompt response sadly the fix addresses only the "it has to be the first and last line" issue.

I suggest changing 
the following new code for https://github.com/duff2013/ulptool/blob/21e08961dd37bd58a5f0ef47707bbc2a5c7ff589/src/esp32ulp_build_recipe.py#L300

```
 if top.startswith("#ifdef _ULPCC_"):
       if bottom.startswith("#endif"):
```

duff2013, 2020-01-09:
Ok I see that looks good do you want to do pull request?

PR #51 (2020-01-09): recipe: make gen_assembly more tolerant
The current source parser is very restrictive any comment would break the compilation. This change improves the situation.

Fixes #50
commit a48ecbaf (2020-01-09): recipe: make gen_assembly more tolerant

The current source parser is very restrictive any comment would
break the compilation. This change improves the situation.

Fixes #50
closed 2020-01-09 by duff2013