duff2013/ulptool #52: Added windows support workaround

enriquewph, 2020-01-28:
Check this out, it's not fancy but it works!

https://github.com/enriquewph/ulptool-windows

duff2013, 2020-01-29:
Nice!, does all the ulpcc examples work? There was bug that I found ([https://github.com/duff2013/lcc/commit/ba41e3b60ca307898256c0f2b40c0eff39c1fd0c](https://github.com/duff2013/lcc/commit/ba41e3b60ca307898256c0f2b40c0eff39c1fd0c)), I assume you fixed that too? There are other cleanups I did, can I see your makefile?

enriquewph, 2020-01-30:
ulp_counter example works fine.

I think i have that same error you described (duff2013/lcc@ba41e3b) too.
I haven't fixed it yet:

> C:\Users\quiqu\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp-elf-binutils/bin/esp32ulp-elf-as -al=adc.ulp.lst -W -o adc.ulp.o adc.ulp.pS
> adc.ulp.pS: Assembler messages:
> 
> 
> adc.ulp.pS:43: Error: syntax error. Input text was 7;.
> 
> 
> adc.ulp.pS:43: Error:

wich is caused by what i think it is the lack of a comma by the adc macro on the generated .s assembly file at line 43:
> adc R0 0,7

I replaced that with:

> adc r0,0,7

and it works just fine.

The way I resolved it to compile on windows is a bit messy but it gets the job done.