duff2013/ulptool #53: esp32 boards location in Windows

mahesh2000, 2020-02-02:
hi, on Windows 10, i found the esp32 board info in `C:\Users\<username>\Documents\ArduinoData\packages\esp32`, not at the location mentioned in the readme file.

duff2013, 2020-02-02:
Did you use the Arduino Board Manager to install the esp32 core? That directory you show is usually where it is installed if you do manual install of the esp32 core.

mahesh2000, 2020-02-02:
I used the boards manager. it's not a big deal, but i thought i'd make a note of it in case someone else had the same issue.

duff2013, 2020-02-02:
I originally used that directory for the ulptool but later switched to the board manager location because that is where most people install the esp32 core. The current iteration of the ulptool will not work in that directory since certain items are not in the same relative location as the board manager install.

mahesh2000, 2020-02-02:
do you mean that i cannot use ulptool since my esp32 core is in `C:\Users\<username>\Documents\ArduinoData\packages\esp32`? as it was automatically put there by the board manager, do i have any way out? thanks.

duff2013, 2020-02-02:
Did you change where the board manager installed it? It's usually in `C:\Users\<USERNAME>\AppData\Local\Arduino15\packages\esp32`

mahesh2000, 2020-02-02:
no, i've not touched it manually. i'm running Arduino 1.8.11 on a Windows 10 machine. is there anything i can do? i don't know it helps, but here's are my preferences:

```
board=generic
boardsmanager.additional.urls=https://dl.espressif.com/dl/package_esp32_index.json, http://arduino.esp8266.com/stable/package_esp8266com_index.json
build.verbose=true
build.warn_data_percentage=75
cache.enable=true
compiler.cache_core=true
compiler.warning_level=none
console=true
console.auto_clear=true
console.error.file=stderr.txt
console.length=500
console.lines=4
console.output.file=stdout.txt
custom_CPUFreq=esp32_240
custom_CrystalFreq=generic_26
custom_DebugLevel=esp32_none
custom_FlashFreq=generic_40
custom_FlashMode=generic_dout
custom_FlashSize=esp32_4M
custom_PSRAM=esp32_disabled
custom_PartitionScheme=esp32_default
custom_ResetMethod=generic_ck
custom_UploadSpeed=esp32_921600
custom_baud=generic_115200
custom_cpu=pro_8MHzatmega328
custom_dbg=generic_Disabled
custom_eesz=generic_512K
custom_exception=generic_disabled
custom_ip=generic_lm2f
custom_led=generic_2
custom_lvl=generic_None____
custom_sdk=generic_nonosdk221
custom_ssl=generic_all
custom_vt=generic_flash
custom_wipe=generic_none
custom_xtal=generic_80
editor.antialias=true
editor.auto_close_braces=true
editor.caret.blink=true
editor.code_folding=false
editor.divider.size=2
editor.external=false
editor.font=Monospaced,plain,12
editor.indent=true
editor.invalid=false
editor.keys.alternative_cut_copy_paste=true
editor.keys.home_and_end_beginning_end_of_doc=false
editor.keys.shift_backspace_is_delete=true
editor.languages.current=
editor.linenumbers=true
editor.save_on_verify=true
editor.tabs.expand=true
editor.tabs.size=2
editor.update_extension=true
editor.window.height.default=600
editor.window.height.min=290
editor.window.width.default=500
editor.window.width.min=400
export.applet.separate_jar_files=false
export.application.fullscreen=false
export.application.platform=true
export.application.stop=true
export.delete_target_folder=true
gui.scale=auto
last.folder=C:\Users\<USERNAME>\Documents\Data\<MAIN_PROJECT_FOLDER>\engg\code\Arduino\Arduino-ESP8266-Secure-Http-Azure-IoT-Hub-Client-V2-master\AzureClient\AzureClient.ino
last.ide.1.8.11.daterun=1580590398
last.ide.1.8.11.hardwarepath=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.28.0_x86__mdqgnx93n4wtt\hardware
[... 41 lines cut ...]
run.options.memory.maximum=256
run.present.bgcolor=#666666
run.present.exclusive=false
run.present.stop.color=#cccccc
serial.databits=8
serial.debug_rate=115200
serial.line_ending=3
serial.parity=N
serial.port=COM18
serial.port.file=COM18
serial.port.iserial=null
serial.show_timestamp=false
serial.stopbits=1
sketchbook.path=C:\Users\<USERNAME>\Documents\Data\<MAIN_PROJECT_FOLDER>\engg\code\Arduino
software=ARDUINO
target_package=esp8266
target_platform=esp8266
theme.file=
update.check=true
update.id=-4713044707227173994
update.last=1580590429282
upload.using=bootloader
upload.verbose=false
upload.verify=true
```

mahesh2000, 2020-02-03:
i kinda got it working in `C:\Users\<username>\Documents\ArduinoData\packages\esp32` by making a few changes:

1. `esp32ulp_mapgen.py` needs to be moved to `C:\Users\<username>\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\tools` (or whatever your version is, instead of `1.0.4`), otherwise error `FileNotFoundError: [WinError 2] The system cannot find the file specified` is generated.
2. You must download and use `binutils-esp32ulp-win32-2.28.51-esp-20191205.zip` as per the instructions, do *not* download `binutils-esp32s2ulp-win32-2.28.51-esp-20191205.zip` as it has a different directory structure.
3. I'm using Python Version 3.8.1, so I had to make a few changes to the file `esp32ulp_build_recipe.py`:
  * changed `print e` to `print (e)`. otherwise Python generated a syntax error.
  * changed `print cmd[0]` to `print (cmd[0])`.
  * changed the file type to `"wb"` instead of `"w"` as i was getting this error: `TypeError:
  write() argument must be str, not bytes`.
  `with open(file_names_constant['sym'],"w") as fsym:`
  to
  `with open(file_names_constant['sym'],"wb") as fsym:`

mahesh2000, 2020-02-03:
**got it working**, i think. if the arduino project does *not* use the ulp, the `esp32\hardware\esp32\1.0.4\platform.local.txt` file must be removed. otherwise while the project compiles, it does not upload. if i figure any of this out, i'll post.

duff2013, 2020-02-11:
Ok I'll check this out maybe they changed something in the esp32 1.0.4 platform.txt?

austinDJ, 2020-12-20:
> i kinda got it working in `C:\Users\<username>\Documents\ArduinoData\packages\esp32` by making a few changes. It still _won't_ upload when ulptool is installed; need to figure out why.
> 
> 1. `esp32ulp_mapgen.py` needs to be moved to `C:\Users\<username>\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\tools` (or whatever your version is, instead of `1.0.4`), otherwise error `FileNotFoundError: [WinError 2] The system cannot find the file specified` is generated. however, the location of this file in the esp32\1.0.4\tools location stops all programs, whether they have ulp code or not, from upload. remove this file and all is good again.
> 2. You must download and use `binutils-esp32ulp-win32-2.28.51-esp-20191205.zip` as per the instructions, do _not_ download `binutils-esp32s2ulp-win32-2.28.51-esp-20191205.zip` as it has a different directory structure.
> 3. I'm using Python Version 3.8.1, so I had to make a few changes to the file `esp32ulp_build_recipe.py`:
> 
> * changed `print e` to `print (e)`. otherwise Python generated a syntax error.
> * changed `print cmd[0]` to `print (cmd[0])`.
> * changed the file type to `"wb"` instead of `"w"` as i was getting this error: `TypeError:   write() argument must be str, not bytes`.
>   `with open(file_names_constant['sym'],"w") as fsym:`
>   to
>   `with open(file_names_constant['sym'],"wb") as fsym:`

This has fixed the same issue I had with 1.0.4. Thank you!