duff2013/ulptool #54: getting compiler error from Python2.7 subprocessor.py

EduardoRenani, 2020-02-06:
OS: macOS Catalina. ESP32 version 1.0.4 installed by Board Manager. Error message:

Traceback (most recent call last):
  File "/Users/eduardoribeiro/Library/Arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 560, in <module>
    main(sys.argv[1:])
  File "/Users/eduardoribeiro/Library/Arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 106, in main
    build_ulp(PATHS, ulp_files, board_options, True)
  File "/Users/eduardoribeiro/Library/Arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 137, in build_ulp
    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 13] Permission denied

duff2013, 2020-02-08:
Hmmm, don't have Catalina on any machines but I know Apple locked down a lot of stuff I wonder if python subsystem calls are a part of that?

duff2013, 2020-02-08:
Are you the Admin of this computer? Can you check the folder ../esp32ulp-elf-binutils/bin/ folder permissions for all the binutils executables?

EduardoRenani, 2020-02-09:
Hey Duff, thanks for the fast reply. I made it work by giving permission for every file inside “../esp32ulp-elf-binutils/bin/” and creating a python virtual environment for Arduino.

closed 2020-02-09 by EduardoRenani