duff2013/ulptool #57: ESP32 ulptool cannot compule example Windows 10/ESP 1.0.4

capolan, 2020-02-19:
Environment:

Windows 10
ESP32 version 1.0.4
Atmel Studio 7 (Version: 7.0.1931 - )
Python 2.7

ESP32 board installed as README.md: C:\Users\capol\AppData\Local\arduino15\packages\esp32\

And successful installed the packages ulptool and esp32ulp-elf-binutils.

Opened the project ulpcc_adc.

The first compiling, occurs the error below:

Building core ...
recipe.hooks.core.postbuild.01.pattern
python "C:\Users\capol\AppData\Local\arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/config" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/app_trace" 
...
bla bla bla
....

 "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/fb_gfx" 
 -b C:\Users\capol\AppData\Local\Temp\VMBuilds\ulpcc_adc\esp32_esp32\Release 
 -p C:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4 
 -u C:\Users\capol\AppData\Local\arduino15\packages\esp32\tools\ulptool\src/esp32ulp-elf-binutils/bin/ 
 -x C:\Users\capol\AppData\Local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/ 
 -t C:\Users\capol\AppData\Local\arduino15\packages\esp32\tools\ulptool\src/ 
 --DF_CPU=240000000L --DARDUINO=10809 --DARDUINO_=ESP32_DEV --DARDUINO_ARCH_=ESP32 --DARDUINO_BOARD="ESP32_DEV" --DARDUINO_VARIANT="esp32"
   Traceback (most recent call last)
   Users\capol\AppData\Local\arduino15\packages\esp32\tools\ulptool\src\esp32ulp_build_recipe.py", line 560, in <module>
   Users\capol\AppData\Local\arduino15\packages\esp32\tools\ulptool\src\esp32ulp_build_recipe.py", line 93, in main
   os.chdir(os.path.join(PATHS[build], sketch))
Error compiling libraries
Build failed for project 'ulpcc_adc'
 
WindowsError*: [Error 2] O sistema não pode encontrar o arquivo especificado: C:\\Users\\capol\\AppData\\Local\\Temp\\VMBuilds\\ulpcc_adc\\esp32_esp32\\Release\\sketch 

Solution: 
edit file esp32ulp_build_recipe.py and include new 2 lines on line 93:

    direc = os.path.join(PATHS['build'], 'sketch')
    if not os.path.exists(direc):
        os.makedirs(direc)
----------------------------------------------------------------------------------
Now, the following error I can not solve.....

Building core ...
recipe.hooks.core.postbuild.01.pattern
python "C:\Users\capol\AppData\Local\arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/config" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/app_trace" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/app_update" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/asio" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/bootloader_support" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/bt" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/coap" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/console" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/driver" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp-tls" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp32" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_event" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_http_client" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_http_server" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/ethernet" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/expat" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/fatfs" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/freemodbus" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/freertos" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/heap" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/idf_test" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/jsmn" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/json" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/libsodium" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/log" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/lwip" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/mbedtls" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/mdns" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/micro-ecc" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/mqtt" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/newlib" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nvs_flash" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/openssl" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/protobuf-c" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/protocomm" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/pthread" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/sdmmc" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/smartconfig_ack" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/soc" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/spi_flash" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/spiffs" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/tcp_transport" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/tcpip_adapter" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/ulp" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/vfs" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/wear_levelling" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/wifi_provisioning" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/wpa_supplicant" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/xtensa-debug-module" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp-face" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware [... cut ...] coexist -lfreemodbus -lspi_flash -lesp-tls -lwpa -lwifi_provisioning -lwps -lesp_adc_cal -lesp_event -lopenssl -lesp_ringbuf -lfr -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\Users\capol\AppData\Local\Temp\VMBuilds\ulpcc_adc\esp32_esp32\Release/ulpcc_adc.ino.elf"

Error linking for board ESP32 Dev Module
Build failed for project 'ulpcc_adc'
 
xtensa-esp32-elf-gcc*: error: C:\Users\capol\AppData\Local\Temp\VMBuilds\ulpcc_adc\esp32_esp32\Release\ulp_main.bin.bin.o: No such file or directory

Thank you.

capolan, 2020-02-19:
Continuing....

the file  **ulp_main.bin.bin.o** is under sketch directory....

C:\Users\capol\AppData\Local\Temp\VMBuilds\ulpcc_adc\esp32_esp32\Release\sketch

$ dir <--- sketch
tmp.ulp.d    tmp.ulp.pS          ulp_main.common.d   ulp_main.h    ulp_main.sym
tmp.ulp.lst  ulp_main.bin        ulp_main.common.ld  ulp_main.ld
tmp.ulp.o    ulp_main.bin.bin.o  ulp_main.elf        ulp_main.map

Build directory:
$ dir
adc.c    adc.c.o   core    sketch      ulpcc_adc.buildtrace  ulpcc_adc.cpp.d
adc.c.d  common.h  core.a  ulp_main.h  ulpcc_adc.cpp         ulpcc_adc.cpp.o

duff2013, 2020-02-19:
windows is not supported yet with the ulp c compiler. Linux and Mac are so far, I'm planning on adding support in the future. Only assembly files will compile using windows.

capolan, 2020-02-21:
Thank you.

Even with the ulp_adc (Assembly) occurs the same error. 

When I switched to another ESP32 project, the same error happened. To compile, i had to remove ulptool.

**A clue:**

In the output window is strange  compiler.c.elf.extra_flags. I do not know where this come from. 

I guess that is related to cache or other stuff of files platform.txt OR platform.local.txt  

version=2.3.0
compiler.ulp.path={runtime.tools.ulptool.path}/esp32ulp-elf-binutils/bin/
tools.ulptool.path={runtime.tools.ulptool.path}/
tools.ulptool.cmd=esp32ulp_build_recipe.py
compiler.s.cmd=python "{tools.ulptool.path}{tools.ulptool.cmd}"
compiler.c.elf.extra_flags="-L{build.path}/" -T ulp_main.ld "{build.path}/ulp_main.bin.bin.o"

-------------------------------------------------------------------------------------------

Building core ...
recipe.hooks.core.postbuild.01.pattern
python "C:\Users\capol\AppData\Local\arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/config" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/app_trace" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/app_update" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/asio" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/bootloader_support" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/bt" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/coap" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/console" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/driver" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp-tls" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp32" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_event" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_http_client" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_http_server" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/ethernet" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/expat" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/fatfs" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/freemodbus" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/freertos" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/heap" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/idf_test" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/jsmn" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/json" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/libsodium" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/log" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/lwip" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/mbedtls" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/mdns" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/micro-ecc" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/mqtt" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/newlib" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nghttp" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/nvs_flash" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/openssl" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/protobuf-c" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/protocomm" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/pthread" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/sdmmc" "-IC:\Users\capol\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/i [... cut ...]  -lcore -lpthread -lcoexist -lfreemodbus -lspi_flash -lesp-tls -lwpa -lwifi_provisioning -lwps -lesp_adc_cal -lesp_event -lopenssl -lesp_ringbuf -lfr -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\Users\capol\AppData\Local\Temp\VMBuilds\ulp_adc\esp32_esp32\Release/ulp_adc.ino.elf"
 

Error linking for board ESP32 Dev Module
xtensa-esp32-elf-gcc*: error: C:\Users\capol\AppData\Local\Temp\VMBuilds\ulp_adc\esp32_esp32\Release\ulp_main.bin.bin.o: No such file or directory
Build failed for project 'ulp_adc'

duff2013, 2020-02-21:
This error you posted says that it didn't find any assembly files (No ULP Assembly File(s) Detected...), does any other ulp examples compile?

capolan, 2020-02-21:
You are right, there is no *.s file in the sketch directory. They are not been copied, I do not know why. 

I add some lines in **esp32ulp_build_recipe.py** to copy *.s and *.c and **works**.

line 88 :

    PATHS = dict()
    PATHS['build']     = args.b
    PATHS['core']      = args.p
    PATHS['ulptool']   = args.t
    PATHS['ucompiler'] = args.u
    PATHS['xcompiler'] = args.x

    direc = os.path.join(PATHS['build'], 'sketch')  // START HERE
    if not os.path.exists(direc):
        os.makedirs(direc)
    os.chdir(direc)

    ulp_files = glob.glob('*.[sc]')
	
    for file in ulp_files:
        try: #deletes the folder
            os.remove(file)
        except OSError:
            pass

    os.chdir('..')
    ulp_files = glob.glob('*.[sc]')
    for file in ulp_files:
        print (file)
        shutil.copy(file,'sketch')
   
    os.chdir(direc)
    ulp_files = glob.glob('*.s')  // ENDS HERE

    gen_assembly(PATHS)