duff2013/ulptool #58: Using example as template results in ULP variables only working 1 way.

bzeeman, 2020-02-20:
```
// initialize ulp variable
  esp_err_t err = ulptool_load_binary(0, ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start) / sizeof(uint32_t));
  // ulp coprocessor will run on its own now
  ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup() );
```

if you initialize ulp variables as shown, accessing those initial values does not seem to work from ULP side of things. 

Example:
```
void setup() {
  Serial.begin(115200);
  delay(500)

  uint32_t wmv = ulp_whats_my_value & 0x0000ffff;
  Serial.printf("Whats My Value: %d\n", wmv);
  init_run_ulp(10 * 10000);
  esp_deep_sleep_start();
}
static void init_run_ulp(uint32_t usec) {

// initialize ulp variable
  ulp_my_value = 42;
  ulp_whats_my_value = 0;
  esp_err_t err = ulptool_load_binary(0, ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start) / sizeof(uint32_t));
  // ulp coprocessor will run on its own now
  ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup() );
  ulp_set_wakeup_period(0, usec);

  err = ulp_run((&ulp_entry - RTC_SLOW_MEM) / sizeof(uint32_t));
  ESP_ERROR_CHECK(ERR);
}
```
with the following assembly code:
```

//proper setup stuff here removed for conciseness 
...
...
entry:
  move r3, my_value
  ld r3, r3, 0 //<-- r3 should now contain the value of my_value

  move r2, whats_my_value
  st r3, r2, 0

  wake
  halt

```
Will print out: "Whats My Value: 0"

However if you instead change the order in init_run_ulp to:
```
static void init_run_ulp(uint32_t usec) {
  esp_err_t err = ulptool_load_binary(0, ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start) / sizeof(uint32_t));
  // ulp coprocessor will run on its own now
  ESP_ERROR_CHECK(err);
  ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup() );
// initialize ulp variable
  ulp_my_value = 42;
  ulp_whats_my_value = 0;

  ulp_set_wakeup_period(0, usec);

  err = ulp_run((&ulp_entry - RTC_SLOW_MEM) / sizeof(uint32_t));
  ESP_ERROR_CHECK(ERR);
}
```

Will print: "Whats My Value: 42"

This only affects 1 direction, if I instead make the assembly:
```

//proper setup stuff here removed for conciseness 
...
...
entry:
  move r3, 42
 
  move r2, whats_my_value
  st r3, r2, 0

  wake
  halt

```

it will print: "Whats My Value: 42"

Also changing the order allows errors when the corresponding variables are not set up correctly in the assembly to be thrown on compile.

duff2013, 2020-02-21:
Right you have to load the binary then initialize your global ulp variables then run your ulp code. I believe this is the same if you use the esp32 IDF instead of Arduino.

bzeeman, 2020-02-21:
I guess al I am saying is that in the read me and comments in examples it seems to indicate it in the wrong order.