duff2013/ulptool #60: ValueError: not enough values to unpack (expected 3, got 0)

Papadoma, 2020-03-14:
ESP32 1.0.4
Latest ulptool

I've fixed some errors on the recipe.py, mostly to do with casting bytes to strings 

Now I get
`python C:\Users\Papadoma\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_mapgen.py -s ulp_main.sym -o ulp_main
Traceback (most recent call last):

  File "C:\Users\Papadoma\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_mapgen.py", line 54, in <module>

    exit(main())

  File "C:\Users\Papadoma\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_mapgen.py", line 49, in main

    gen_ld_h_from_sym(f_sym, f_ld, f_h)

  File "C:\Users\Papadoma\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_mapgen.py", line 22, in gen_ld_h_from_sym

    name, _, addr_str = line.split()

ValueError: not enough values to unpack (expected 3, got 0)`

duff2013, 2020-07-08:
Can you give me link to your fixes?

Papadoma, 2020-07-08:
I haven't worked on it for a while
I will give it another look this week and come back to you

JosueAsturias, 2020-11-25:
i have this problem in windows 10, how can it be solved?

JosueAsturias, 2020-11-25:
Traceback (most recent call last):l\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_mapgen.py -s ulp_main.sym -o ulp_main
  File "C:\Users\jotas\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_mapgen.py", line 54, in <module>
    exit(main())
  File "C:\Users\jotas\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_mapgen.py", line 49, in main
    gen_ld_h_from_sym(f_sym, f_ld, f_h)
  File "C:\Users\jotas\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_mapgen.py", line 22, in gen_ld_h_from_sym
    name, _, addr_str = line.split()
ValueError: not enough values to unpack (expected 3, got 0)

exit status 1
Error compiling for board ESP32 Dev Module.

rheingoldheavy, 2022-09-20:
Here in Sept-2022 this problem also occurred for me when attempting to build any of the example code I've found. 

Environment:

- Windows 10
- Arduino IDE 1.8.13
- Python 3.9-64 (default)
- Python 3.88-32
- Python 2.7-64
- ulptool v2.4.2

The steps I took to get here (not sure of the order)...

1. Full ulptool installation procedure followed to the letter
2. Added `# python -2` to the top of both esp32ulp_build_recipe.py and esp32ulp_mapgen.py, to force the use of Python v2.x
3. Fix "TypeError: can only concatenate str" issue using this post: [Issue #62 ](https://github.com/duff2013/ulptool/issues/62)
4. Fix "undefined reference" due to not copying .s files: [60d0f7d](https://github.com/duff2013/ulptool/commit/60d0f7d59439179ebf3e5e879ce4abcde37e4d3b)

At that point I was left with the "ValueError: not enough values to unpack" issue. I tracked that down to the generation of the symbol "ulp_main.sym" file performed in esp32ulp_build_recipe.py using [nm](https://en.wikipedia.org/wiki/Nm_(Unix)) resulting in extra empty lines. When esp32ulp_mapgen.py attempts to parse the line and split it using space as a delimiter, it doesn't know what to do with the empty lines and errors out.

I hacked together a fix for esp32ulp_build_recipe.py using this [StackOverflow post](https://codereview.stackexchange.com/questions/145126/open-a-text-file-and-remove-any-blank-lines) as guidance.

**Original Code**
`    ## Generate list of global symbols
    cmd = gen_binutils_nm_cmd(PATHS)
    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)
    (out, err) = proc.communicate()
    if err:
        error_string = cmd[0] + '\r' + err
        sys.exit(error_string)
    else:
        file_names_constant = gen_file_names_constant()
        with open(file_names_constant['sym'],"w") as fsym:
            fsym.write(out.decode('utf-8'))
        console_string += cmd[0] + '\r'
`

**Revised Code**
`    ## Generate list of global symbols
    cmd = gen_binutils_nm_cmd(PATHS)
    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)
    (out, err) = proc.communicate()
    if err:
        error_string = cmd[0] + '\r' + err.decode('utf-8')
        sys.exit(error_string)
    else:
        file_names_constant = gen_file_names_constant()
        
        with open(file_names_constant['sym'],"w") as fsym:
            fsym.write(out.decode('utf-8'))

        with open(file_names_constant['sym']) as filehandle:
            lines = filehandle.readlines()

        with open(file_names_constant['sym'], 'w') as filehandle:
            lines = filter(lambda x: x.strip(), lines)
            filehandle.writelines(lines)   

        console_string += cmd[0] + '\r'
`

I've successfully compiled and run the readme example, and compiled a handful of the other ulptool examples (haven't tried flashing them yet, but they compile).  

Hopefully someone finds this helpful :)