duff2013/ulptool #61: Strange issue when linking multiple .s files

fededim, 2020-04-24:
Hi,

I am attaching [this ](https://github.com/duff2013/ulptool/files/4527163/ulp_test.zip) arduino project.

Tha main ulp.s file justs wakes up the ESP32 and halts (almost everything is commented). I had to increase the CONFIG_ULP_COPROC_RESERVE_MEM in sdkconfig.h since the binary assembly is more than 512 bytes. The problem is that the project does not work on esp32, it stays foreover in deep sleep mode without being awaken.

If instead you delete the i2c.s file and recompile, everything works as expected (ESP32 is awaken up every 5 seconds). **I want to highline that in i2c.s there are only i2c subroutines which are not used by main ulp.s (e.g. they are not called).**.

I believe that there is a problem with the linking process which seems not to work well with multiple .s files. There is also the stack.s file, but it contains only macros so it should not contribute to the final binary code. Can you do a check and let me know ?

pikot, 2020-07-05:
i have same issue on linux, but all work fine on mac

more information, i use

- arduino ide 1.8.12,
- binutils-esp32ulp-macos-2.28.51-esp-20191205
- ulptool-2.4.1
- maxos / linux (same setup, only binutils for appropriate os)

i created small test-case repo, where i out via ulp_debug code, and i understand why it's happened, but i don't have any idea how to fix

 https://github.com/pikot/ulp_main_test, 

here is  problem output, on linux mashine

00:28:07.062 -> ====================================
00:28:07.062 -> 0000 : 80200001 PROG JUMP   R1			// GOTO R1
00:28:07.095 -> 0001 : 72800081 PROG MOVE   R1,   8			// R1 = 8
00:28:07.095 -> 0002 : 72800020 PROG MOVE   R0,   2			// R0 = 2
00:28:07.095 -> 0003 : 68000004 PROG ST     R0, R1,   0			// MEM[R1+0] = R0
00:28:07.095 -> 0004 : B0000000 PROG HALT				// HALT
00:28:07.095 -> iter 0; 

here is ok, on mac

0000 : 72800061 PROG MOVE   R1,   6                      // R1 = 6
0001 : 72800020 PROG MOVE   R0,   2      // R0 = 2
0002 : 68000004 PROG ST     R0, R1,   0                 // MEM[R1+0] = R0
0003 : B0000000 PROG HALT                              // HALT
0004 : 80200001 PROG JUMP   R1                      // GOTO R1
0005 : 00000000 PROG NOP                              // NOP
0006 : 00410002 DATA     2                                // ST ADDR:0x0002
iter 2;

as we may see, here  command "80200001 PROG JUMP   R1                      // GOTO R1" inserted as first command, and it is a root of the problem, as i understand this is a linker problem?

pikot, 2020-07-05:
i guess this pull request help for this  fix-error,
https://github.com/duff2013/ulptool/pull/55
at least for me =)

duff2013, 2020-07-08:
I'm currently working on another project now but will be back with ulptool soon.

draget, 2020-11-13:
> i guess this pull request help for this fix-error,
> #55
> at least for me =)

same here - works