duff2013/ulptool #62: fsys.write() expecting string but parameter is byte

swaraj429, 2020-05-11:
platform:  linux
python 2.7.16
Traceback (most recent call last):
  File "/home/swaraj/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 503, in <module>
    main(sys.argv[1:])
  File "/home/swaraj/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 102, in main
    build_ulp(PATHS, ulp_files, board_options, True)
  File "/home/swaraj/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 230, in build_ulp
    fsym.write(out)
TypeError: write() argument must be str, not bytes
exit status 1
Error compiling for board ESP32 Dev Module.

PauloFernandoSF, 2020-05-29:
I have a similar error.
Platform:linux
Python:3.7.7
Esp core:1.0.4

Traceback (most recent call last):
  File "/home/paulo/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 560, in <module>
    main(sys.argv[1:])
  File "/home/paulo/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 106, in main
    build_ulp(PATHS, ulp_files, board_options, True)
  File "/home/paulo/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 160, in build_ulp
    error_string = cmd[0] + '\r' + err
TypeError: can only concatenate str (not "bytes") to str

duff2013, 2020-05-29:
@swaraj429 I'm not sure whats the issue can you post the whole error you are seeing?

@PauloFernandoSF only python 2.7 is supported currently, can you use that instead and report back any errors?

PauloFernandoSF, 2020-05-29:
Thanks, i've changed do Python 2.7 and worked!

O1dAlex, 2020-05-30:
I've modified script - now it works with python 3.

This particular error corrected in line 233 - file 'fsym' should be open in binary mode:
`        with open(file_names_constant['sym'],"wb") as fsym:
`

swaraj429, 2020-06-01:
> @swaraj429 I'm not sure whats the issue can you post the whole error you are seeing?
> @PauloFernandoSF only python 2.7 is supported currently, can you use that instead and report back any errors?

that's all the error I got

2010019970909, 2020-11-13:
@swaraj429,

In the `platform.local.txt` (read the readme to find its location), could you change `python` to `python2`:

```
## ulp build tool
compiler.s.cmd=python "{tools.ulptool.path}{tools.ulptool.cmd}"
``` 
to:
```
## ulp build tool
compiler.s.cmd=python2 "{tools.ulptool.path}{tools.ulptool.cmd}"
``` 

I think in the newest Linux version, `python` is actually Python 3.

Regards,
Vincent

nightlyteacups, 2021-03-27:
Having the same issue as @PauloFernandoSF had right now. Python 3.8.5, 1.0.6 core, Arduino 1.8.13, Win. Gonna downgrade Python to 2.7 and check again.

yeckel, 2021-05-05:
I'm observing a similar issue when there is an ASM compile issue. Like typo or other problem. The compiler is not reporting where and what bug you have in the code, but instead: _TypeError: can only concatenate str (not "bytes") to str_

2010019970909, 2021-05-05:
> I'm observing a similar issue when there is an ASM compile issue. Like typo or other problem. The compiler is not reporting where and what bug you have in the code, but instead: _TypeError: can only concatenate str (not "bytes") to str_

It's probably the still issue. Which version of Python are you using? And can you paste the full error code?

yeckel, 2021-05-05:
Python 3.8.5 at Ubuntu 20.04, I've added "hi" into the ASM code:
```
Traceback (most recent call last):
  File "/home/libor/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 560, in <module>
    main(sys.argv[1:])
  File "/home/libor/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 106, in main
    build_ulp(PATHS, ulp_files, board_options, True)
  File "/home/libor/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 140, in build_ulp
    error_string = cmd[0] + '\r' + err
TypeError: can only concatenate str (not "bytes") to str
exit status 1
Error compiling for board TTGO LoRa32-OLED V1.
```

2010019970909, 2021-05-05:
Can you change the `err` to `err.decode('utf-8')` at the line 40 in the file `"/home/libor/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py"`?
From what I see, `err` is a bytes array.

yeckel, 2021-05-05:
Sure, and it helped!
```
adc.ulp.pS: Assembler messages:/esp32/tools/ulptool/src/esp32ulp-elf-binutils/bin/esp32ulp-elf-as -al=adc.ulp.lst -W -o adc.ulp.o adc.ulp.pS
adc.ulp.pS:19: Error: syntax error. Input text was hi.
adc.ulp.pS:19: Error: 

exit status 1
Error compiling for board ESP32 Wrover Module.
```
probably all 4 remaining ```error_string = cmd[0] + '\r' + err``` could be changed too.

2010019970909, 2021-05-05:
Glad it helped. And yes, the remaining lines can be changed.