duff2013/ulptool #63: cannot compile examples on linux

wodz, 2020-09-02:
arduino 1.8.13, ulptool 2.4.1, binutils-esp32ulp-linux-amd64-2.28.51-esp-20191205
Everything setup according to readme
**1) ulp_adc**
Arduino: 1.8.13 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

/home/wodz/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/wodz/arduino-1.8.13/hardware -hardware /home/wodz/.arduino15/packages -tools /home/wodz/arduino-1.8.13/tools-builder -tools /home/wodz/arduino-1.8.13/hardware/tools/avr -tools /home/wodz/.arduino15/packages -built-in-libraries /home/wodz/arduino-1.8.13/libraries -libraries /home/wodz/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10813 -build-path /tmp/arduino_build_160625 -warnings=none -build-cache /tmp/arduino_cache_158769 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/home/wodz/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/home/wodz/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.mkspiffs.path=/home/wodz/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/wodz/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/wodz/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/home/wodz/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -verbose /home/wodz/Arduino/ulp_examples/ulp_adc/ulp_adc.ino
/home/wodz/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/wodz/arduino-1.8.13/hardware -hardware /home/wodz/.arduino15/packages -tools /home/wodz/arduino-1.8.13/tools-builder -tools /home/wodz/arduino-1.8.13/hardware/tools/avr -tools /home/wodz/.arduino15/packages -built-in-libraries /home/wodz/arduino-1.8.13/libraries -libraries /home/wodz/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10813 -build-path /tmp/arduino_build_160625 -warnings=none -build-cache /tmp/arduino_cache_158769 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/home/wodz/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/home/wodz/.arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.mkspiffs.path=/home/wodz/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/wodz/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/wodz/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/home/wodz/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -verbose /home/wodz/Arduino/ulp_examples/ulp_adc/ulp_adc.ino
Using board 'esp32' from platform in folder: /home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4
Using core 'esp32' from platform in folder: /home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4
Detecting libraries used...
/home/wodz/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32 -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_server -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_https_ota -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_ringbuf -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ethernet -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/expat -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fatfs -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freemodbus -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freertos -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/heap -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/idf_test -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/jsmn -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/json -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/libsodium -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/log -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/lwip -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mbedtls -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mdns -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/micro-ecc -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mqtt -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/newlib -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nghttp -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nvs_flash -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/openssl -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protobuf-c -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protocomm -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/pthread -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/sdmmc -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/smartconfig_ack -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/soc -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spi_flash -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spiffs -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcp_transport -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcpip_adapter -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ulp -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/vfs -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wear_levelling -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wifi_provisioning -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wpa_supplicant -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/xtensa-debug-module -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32-camera -I/home/wodz/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sd [... cut ...]
b"/home/wodz/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp-elf-binutils/bin/esp32ulp-elf-ld: ulp_main.elf section `.text' will not fit in region `ram'\n/home/wodz/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp-elf-binutils/bin/esp32ulp-elf-ld: region `ram' overflowed by 608 bytes\n"
exit status 1
Error compiling for board ESP32 Dev Module.

**6) ulp_rtc_gpio**
compiles ok

**7) ulp_tsens**
compiles ok

**8) ulp_watering_device**
compiles ok

Errors seems to come in two flavours - not enough ram for ulp program and ulp symbols not defined. Any hint how to compile examples (I am interested in ulpcc particularly)?

duff2013, 2021-01-23:
You need to extend the useable amount of rtc ram seen by the ulp. 

Try to edit the sdkconfig file in your esp32-arduino install. Search for CONFIG_ULP_COPROC_RESERVE_MEM and add more memory like 1024. My sdkcofig file was located in ../tools/sdk/

angyongen, 2021-08-06:
my sdkconfig was in `tools\sdk\include\config\sdkconfig.h`