duff2013/ulptool #65: Problems to compile ULP examples Win 10

vesatam, 2020-09-11:
I think I got quite close with this interesting ulptool. I tried fully follow your instructions. However my limited expertise stopped to  error listings below. Can you help?
I tried to compile various ULP examples.
Windows 10
Python3.7
Arduino IDE 1.8.13

.....
Compiling libraries...

Compiling core...

Using precompiled core: C:\Users\vesa_\AppData\Local\Temp\arduino_cache_726366\core\core_ff867e9ab0d69d87c5bd4b90d4435fb8.a

python "C:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\ulptool\\src/esp32ulp_build_recipe.py" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nghttp" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nvs_flash" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/openssl" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protobuf-c" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protocomm" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/pthread" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/sdmmc" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/smartconfig_ack" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/soc" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/spi_flash" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/spiffs" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/tcp_transport" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/tcpip_adapter" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ulp" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/vfs" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wear_levelling" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wifi_provisioning" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wpa_supplicant" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/xtensa-debug-module" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-face" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32-camera" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-face" "-IC:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fb_gfx" -b "C:\\Users\\vesa_\\AppData\\Local\\Temp\\arduino_build_936925" -p "C:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4" -u "C:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\ulptool\\src/esp32ulp-elf-binutils/bin/" -x "C:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/" -t "C:\\Users\\vesa_\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\ulptool\\src/" --DF_CPU=240000000L --DARDUINO=10813 --DARDUINO_=ESP32_DEV --DARDUINO_ARCH_=ESP32 "--DARDUINO_BOARD=\"ESP32_DEV\"" "--DARDUINO_VARIANT=\"esp32\""

Traceback (most recent call last):

  File "C:\Users\vesa_\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 560, in <module>

    main(sys.argv[1:])

  File "C:\Users\vesa_\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 106, in main

    build_ulp(PATHS, ulp_files, board_options, True)

  File "C:\Users\vesa_\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 137, in build_ulp

    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)

  File "C:\Users\vesa_\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 775, in __init__

    restore_signals, start_new_session)

  File "C:\Users\vesa_\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 1178, in _execute_child

    startupinfo)

FileNotFoundError: [WinError 2] The system cannot find the file specified

exit status 1

Error compiling for board ESP32 Dev Module.

juanhuarte, 2020-10-26:
i have the same problem, could you solve it?

vesatam, 2020-10-26:
Hi, unfortunately I couldn’t find solution this way. I changed my whole concept.

duff2013, 2020-10-26:
Make sure your binutils is the right version: [https://github.com/espressif/binutils-esp32ulp/releases/latest](https://github.com/espressif/binutils-esp32ulp/releases/latest) and not the esp32s2 version. 

It needs to be installed in this directory: "C:\Users\vesa_\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/"

juanhuarte, 2020-11-02:
i try again and this happened
File "C:\Users\juan_\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src/esp32ulp_build_recipe.py", line 305
    print e
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(e)?
exit status 1
Error compiling for board ESP32 Wrover Module.

duff2013, 2020-11-02:
it's because I wrote it in python 2.7, can you use that?

savagerex, 2021-02-23:
How to install the code (https://github.com/espressif/binutils-esp32ulp/releases/latest)  ???

ahosny333, 2021-02-25:
i also havethe same problem , anyone can solve it 

Traceback (most recent call last):

  File "C:\Users\Laptop\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src\esp32ulp_build_recipe.py", line 560, in <module>

    main(sys.argv[1:])

  File "C:\Users\Laptop\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src\esp32ulp_build_recipe.py", line 106, in main

    build_ulp(PATHS, ulp_files, board_options, True)

  File "C:\Users\Laptop\AppData\Local\Arduino15\packages\esp32\tools\ulptool\src\esp32ulp_build_recipe.py", line 137, in build_ulp

    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)

  File "C:\Users\Laptop\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 947, in __init__

    self._execute_child(args, executable, preexec_fn, close_fds,

  File "C:\Users\Laptop\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 1416, in _execute_child

    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,

FileNotFoundError: [WinError 2] The system cannot find the file specified

exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

savagerex, 2021-02-25:
please follow this step

Download the latest release of this repository and unpack-> https://github.com/duff2013/ulptool/releases/latest delete the release version number so the folder is just called 'ulptool'

Download and unpack the latest pre-compiled binutils-esp32ulp toolchain for Mac/Linux/Windows: https://github.com/espressif/binutils-esp32ulp/releases/latest

Find your Arduino-esp32 core directory which Arduino IDE uses:

Typically (Mac OS) -> ~/Library/Arduino15/packages/esp32

Typically (Windows) -> C:\Users<USERNAME>\AppData\Local\Arduino15\packages\esp32

Typically (Linux) -> ~/.arduino15/packages/esp32

Move the ulptool folder you downloaded and unpacked to the tools folder here -> ... /esp32/tools/ulptool/.

Copy the 'platform.local.txt' file to ... /esp32/hardware/esp32/1.0.0/. Remember 1.0.0 has to match your esp32 core version.

In the ulptool folder, move or copy the ... /ulptool/src/ulp_examples folder to where Arduino saves your sketches.

Move esp32ulp-elf-binutils folder you downloaded and unpacked to -> ... /esp32/tools/ulptool/src/esp32ulp-elf-binutils/.

ahosny333, 2021-02-26:
@savagerex thanks for your replay but i already have followed this steps and same error , i am using Python 3.9.0 
is that python version ok or i must use python 2.7 ?

savagerex, 2021-02-26:
Yes, please use python 2.7

ahosny333, 2021-02-26:
i am using windows 10, have any steps to setup python 2.7 with python 3.9.0 ?

savagerex, 2021-02-26:
you must uninstall python3.9.0, then install python 2.7

Jorge-spicymexican, 2021-06-23:
when you try to run python 2.7 you just get this error instead.
exec: "python": executable file not found in %PATH%
Error compiling for board ESP32 Dev Module.
I believe its because pything 2.7 is no longer being supported by audrinos ide

mahesh2000, 2021-07-27:
using the correct version of esp32ulp-elf-binutils, not the one with s2 in the name, is one of the problems I had to fix before getting things working.