duff2013/ulptool #66: exec: "python": executable file not found in %PATH%    Error compiling for board WEMOS LOLIN32.

jay3x7, 2021-01-01:
Followed the directions to install on the front page. I believe that all was done correctly. 
Found in comments that we needed to install python 2.7. Installed python 2.7.18.
Donwgraded boards manager to 1.0.2. Saw in comments that this should be supported.
Did not see anywhere on how to link python to arduino sketches?
Happy New Year, and thank you!

IDE outputs this:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Dell\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Dell\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Dell\Documents\Arduino\libraries -fqbn=esp32:esp32:lolin32:FlashFreq=80,PartitionScheme=default,UploadSpeed=921600 -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\Dell\AppData\Local\Temp\arduino_build_531931 -warnings=default -build-cache C:\Users\Dell\AppData\Local\Temp\arduino_cache_750718 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\Dell\Documents\Arduino\ulp_examples\ulp_adc\ulp_adc.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Dell\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Dell\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Dell\Documents\Arduino\libraries -fqbn=esp32:esp32:lolin32:FlashFreq=80,PartitionScheme=default,UploadSpeed=921600 -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\Dell\AppData\Local\Temp\arduino_build_531931 -warnings=default -build-cache C:\Users\Dell\AppData\Local\Temp\arduino_cache_750718 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -verbose C:\Users\Dell\Documents\Arduino\ulp_examples\ulp_adc\ulp_adc.ino
Using board 'lolin32' from platform in folder: C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Using core 'esp32' from platform in folder: C:\Users\Dell\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Detecting libraries used...
"C:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/config" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_trace" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_update" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/asio" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bootloader_support" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bt" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/coap" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/console" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/driver" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp-tls" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp32" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_event" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_http_client" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_http_server" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_https_ota" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/ethernet" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/expat" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/fatfs" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/freemodbus" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/freertos" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/heap" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/idf_test" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/jsmn" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/json" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/libsodium" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/log" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/lwip" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mbedtls" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mdns" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/micro-ecc" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mqtt" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/newlib" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/nghttp" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/nvs_flash" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/openssl" "-IC:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/protobuf-c" "-IC:\\Users\\Dell\\AppData\\Local\\A [... cut ...] 32\\tools\\ulptool\\src/esp32ulp-elf-binutils/bin/" -x "C:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/" -t "C:\\Users\\Dell\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\ulptool\\src/" --DF_CPU=240000000L --DARDUINO=10813 --DARDUINO_=LOLIN32 --DARDUINO_ARCH_=ESP32 "--DARDUINO_BOARD=\"LOLIN32\"" "--DARDUINO_VARIANT=\"lolin32\""
exec: "python": executable file not found in %PATH%
Error compiling for board WEMOS LOLIN32.

jay3x7, 2021-01-01:
UPDATE:

Upon further looking around I discovered the post in your issues area.
Went to:  https://www.pythoncentral.io/add-python-to-path-python-is-not-recognized-as-an-internal-or-external-command/
Followed the instructions and it worked.
Please close this.

BroellLgg, 2021-02-17:
Hi, 

I have the same issue. 
I am on Windows 10 (64Bit), using Arduino 1.8.13 
Installed Python 2.7 (2.7.12) 
added the path following this instruction: 
https://www.pythoncentral.io/add-python-to-path-python-is-not-recognized-as-an-internal-or-external-command/

got it running on cmd: 

[image]

tried it with esp32 board manager versions 1.0.0, 1.0.1, and 1.0.4 
pasted the platform.local.txt to ... /esp32/hardware/esp32/1.0.0/  or .../1.0.1/ or /1.0.4/

and also tried it with Python 3.9 (and working cmd path) 

and everytime i get this error: 
exec: "python": executable file not found in %PATH%
Fehler beim Kompilieren für das Board Adafruit ESP32 Feather.

Has anyone any idea what else could be the reason for this Problem?

BroellLgg, 2021-02-17:
UPDATE: 

I uninstalled Arduino and installed it again and now it is working, i have no idea why...

Jorge-spicymexican, 2021-06-23:
what audrino version did you installed by chance?

luqman8230339900, 2021-08-19:
I'm trying to install duff2013 ulptool from https://github.com/duff2013/ulptool. After did all the steps as mention on that link but when I execute my code it shows an error

"
exec: "python": executable file not found in %PATH% Error compiling for board ESP32 Dev Module.

my directories, ulptool: C:\Users\7896\Documents\ArduinoData\packages\esp32\tools\ulptool

esp32ulp-elf-binutils: C:\Users\7896\Documents\ArduinoData\packages\esp32\tools\ulptool\src\esp32ulp-elf-binutils

and third platform.local.txt C:\Users\7896\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\platform.local.txt

What I got when I put that text file all the examples of Arduino ide shows python path error which I mention above when I remove that file from specific place again it works right.

what I want to achieve: I want to run uLPtool examples UlpAdc.
i also add pyhton3.8 path in environmental variables but still not helping

IAMAHGK, 2023-02-16:
> I'm trying to install duff2013 ulptool from https://github.com/duff2013/ulptool. After did all the steps as mention on that link but when I execute my code it shows an error
>
> " exec: "python": [...]

Hey @luqman8230339900 did you found a solution I have been also trying to run UlpAdc.

luqman8230339900, 2023-02-16:
What exactly you're trying to do can you explain what duf2013?
What you want to run the esp32 kit on Arduino?

IAMAHGK, 2023-02-17:
> What exactly you're trying to do can you explain what duf2013? What you want to run the esp32 kit on Arduino?
> […](#)

@luqman8230339900 I have been trying to read values from the Esp32 pins when the Esp32 is in deep sleep mode but I have been facing problems with this code
I get errors like either:
Undefined reference
Python executable file not found