duff2013/ulptool #71: Undefined reference

jbrepogmailcom, 2021-02-24:
Hello, I am trying to compile but getting an error. I had to move ulptool.h to example folder and I also increased ULP memory to 1024. That did not chnge the error I am still getting:
```
/tmp/arduino_build_153116/sketch/ulp_test.ino.cpp.o:(.literal._Z5setupv+0x10): undefined reference to `ulp_count'
/tmp/arduino_build_153116/sketch/ulp_test.ino.cpp.o:(.literal._Z5setupv+0x18): undefined reference to `_binary_ulp_main_bin_end'
/tmp/arduino_build_153116/sketch/ulp_test.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `_binary_ulp_main_bin_start'
/tmp/arduino_build_153116/sketch/ulp_test.ino.cpp.o:(.literal._Z5setupv+0x20): undefined reference to `ulp_entry'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Wrover Module.
```

savagerex, 2021-02-25:
please follow this step

1. Download the latest release of this repository and unpack-> https://github.com/duff2013/ulptool/releases/latest delete the release version number so the folder is just called 'ulptool'

2. Download and unpack the latest pre-compiled binutils-esp32ulp toolchain for Mac/Linux/Windows: https://github.com/espressif/binutils-esp32ulp/releases/latest

3. Find your Arduino-esp32 core directory which Arduino IDE uses:

     Typically (Mac OS) -> ~/Library/Arduino15/packages/esp32

     Typically (Windows) -> C:\Users\<USERNAME>\AppData\Local\Arduino15\packages\esp32

     Typically (Linux) -> ~/.arduino15/packages/esp32
4. Move the ulptool folder you downloaded and unpacked to the tools folder here -> ... /esp32/tools/ulptool/.

5. Copy the 'platform.local.txt' file to ... /esp32/hardware/esp32/1.0.0/. Remember 1.0.0 has to match your esp32 core version.

6. In the ulptool folder, move or copy the ... /ulptool/src/ulp_examples folder to where Arduino saves your sketches.

7. Move esp32ulp-elf-binutils folder you downloaded and unpacked to -> ... /esp32/tools/ulptool/src/esp32ulp-elf-binutils/.

jbrepogmailcom, 2021-02-25:
Hello. Did the process exactly same way twice and upgraded arduino IDE to 1.8.13. Now I am getting different error:
```
Traceback (most recent call last):
  File "/home/janbenes/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 560, in <module>
    main(sys.argv[1:])
  File "/home/janbenes/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 106, in main
    build_ulp(PATHS, ulp_files, board_options, True)
  File "/home/janbenes/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 137, in build_ulp
    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
exit status 1
Error compiling for board ESP32 Dev Module.
```

jbrepogmailcom, 2021-02-25:
Tried to rename esp32s2ulp-elf-binutils to esp32ulp-elf-binutils and esp32s2ulp-elf to esp32ulp-elf but it did not help

ironmanrex, 2021-02-25:
these error, i ever meet.

you just follow there step, it will be success.

check every step is correct.

thx!!

jbrepogmailcom, 2021-02-25:
Sorry did not work. Just to be sure - in your guide "Copy the 'platform.local.txt' file to ... /esp32/hardware/esp32/1.0.0/. Remember 1.0.0 has to match your esp32 core version." - means copy the file from where?

Jan

ironmanrex, 2021-02-25:
you can choose 1.0.4.

i try it . it will success

jbrepogmailcom, 2021-02-25:
Hello, the differences are: I have folder called 1.0.5  and in your guide step 7, I have name esp32s2ulp-elf-binutils, not esp32ulp-elf-binutils, because esp32s2ulp-elf-binutils was the file I downloaded from github.

jbrepogmailcom, 2021-02-25:
OK, so I have actually had to download different file, binutils-esp32ulp-linux-amd64-2.28.51-esp-20191205.tar.gz .

But after compiling, I get following error:

/home/janbenes/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp-elf-binutils/bin/esp32ulp-elf-ld: ulp_main.elf section `.data' will not fit in region `ram'_main.common.ld adc.ulp.o
/home/janbenes/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp-elf-binutils/bin/esp32ulp-elf-ld: region `ram' overflowed by 88 bytes

Does not matter if I set CONFIG_ULP_COPROC_RESERVE_MEM=1024 or CONFIG_ULP_COPROC_RESERVE_MEM=4096, still by 88 bytes...

savagerex, 2021-02-26:
try increasing the ulp ram by finding CONFIG_ULP_COPROC_RESERVE_MEM in sdkconfig.h

CONFIG_ULP_COPROC_RESERVE_MEM 1024

jbrepogmailcom, 2021-02-26:
I just wrote in my previous post that I have done that as well...

savagerex, 2021-02-26:
OK!!

mahesh2000, 2021-07-27:
manually copying over the .s files to the temp folder works. see https://github.com/duff2013/ulptool/issues/78.