duff2013/ulptool #73: ULP in Eclipse CDT

dmarc1234, 2021-06-09:
I am trying to get the ulptool working under Eclipse CDT.

I have followed the Arduino IDE install procedure placing the relevant files within the CDT tools directory, the ULP compiler under src/ and the platform.local.txt in the same location as the platform.txt file but the path information does not seem to be right as I get the following errors during compile:

project/main.cpp.o:(.literal._Z4loopv+0x7c): undefined reference to `_binary_ulp_main_bin_end'
project/main.cpp.o:(.literal._Z4loopv+0x80): undefined reference to `_binary_ulp_main_bin_start'
project/main.cpp.o:(.literal._Z4loopv+0x90): undefined reference to `ulp_low_thr'
project/main.cpp.o:(.literal._Z4loopv+0x94): undefined reference to `ulp_high_thr'
project/main.cpp.o:(.literal._Z4loopv+0x9c): undefined reference to `ulp_sample_counter'
project/main.cpp.o:(.literal._Z4loopv+0xa0): undefined reference to `ulp_entry'
collect2.exe: error: ld returned 1 exit status

Any idea if your ulptool will work in this environment, any pointers would be much appreciated.

David

dmarc1234, 2021-06-24:
Just wondering if anyone is going to help with this ?

dmarc1234, 2021-07-06:
It would be great if someone could help with this, anyone ?

David

angyongen, 2021-08-06:
maybe #78 helps?