duff2013/ulptool #74: Cannot compile README example: undefined reference

TiborVoelcker, 2021-06-13:
Hi all and @duff2013.
You must get tired of these, I know. I saw a lot of them in the issues. But I went trough all of them and could not find a solution. I did follow the installation manual precisely and also did it twice.
When compiling the ulp_README example, I get following error:
´´´
sketch\ulp_README.ino.cpp.o:(.literal._Z5setupv+0x18): undefined reference to `ulp_count'
sketch\ulp_README.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `ulp_entry'
collect2.exe: error: ld returned 1 exit status
exit status 1
´´´
OS: Windows 10 x64
Arduino IDE: 1.8.15
ulptool: 2.4.1
binutils: binutils-esp32ulp-win32-2.28.51-esp-20191205
board: 1.0.0

My files:
Arduino15/esp32/
├───hardware
│   └───esp32
│       └───1.0.1
│           ├───cores/
│           ├───libraries/
│           ├───tools/
│           ├───variants/
│           ├───boards.txt
│           ├───*platform.local.txt*
│           ├───platform.txt
│           └───programmers.txt
└───tools
    ├───esptool_py/
    ├───mkspiffs/
    ├───*ulptool/*
    │   ├───src/
    │   │  ├───*esp32ulp-elf-binutils/*
    │   │  ├───include/
    │   │  ├───ld/
    │   │  ├───ulpcc/
    │   │  ├───esp32ulp_build_recipe.py
    │   │  └───ld
    │   ├───platforms.local.txt
    │   ├───README.md
    │   └───revisions.md
    └───xtensa-esp32-elf-gcc/

ceerqingtingml, 2021-06-22:
@TiborVoelcker I have the same issue with you, I am wondering have you resolved this issue?

TiborVoelcker, 2021-06-22:
@ceerqingtingml Sadly I could not. I am pretty sure its just a mistake in my setup, but I couldn't get it to work. I switched to VS Code and the ESP-IDF.

gerardpalk, 2021-06-22:
Hi.  I am also having a similar issue.  I have had this library work previously. But my previous PC was fried by a lightning strike.  I am just tryign to rebuild everything, and cant get the ulptool library to work: 

sketch\weather_esp32_main_1.ino.cpp.o:(.literal._Z5setupv+0x20): undefined reference to `ulp_debounce_counter'
sketch\weather_esp32_main_1.ino.cpp.o:(.literal._Z5setupv+0x24): undefined reference to `ulp_debounce_max_count'
sketch\weather_esp32_main_1.ino.cpp.o:(.literal._Z5setupv+0x28): undefined reference to `ulp_next_edge'
sketch\weather_esp32_main_1.ino.cpp.o:(.literal._Z5setupv+0x2c): undefined reference to `ulp_io_number'
sketch\weather_esp32_main_1.ino.cpp.o:(.literal._Z5setupv+0x34): undefined reference to `ulp_edge_count_to_wake_up'
sketch\weather_esp32_main_1.ino.cpp.o:(.literal._Z5setupv+0x40): undefined reference to `ulp_entry'
sketch\weather_esp32_main_1.ino.cpp.o:(.literal._Z5setupv+0x50): undefined reference to `ulp_edge_count'

gerardpalk, 2021-06-23:
I have the same errors as @TiborVoelcker when trying to compile ulp_README example.

gerardpalk, 2021-06-26:
@ceerqingtingml - did you manage to sort this issue?  I still havent worked out the problem.  I am running:
OS: Windows 10 Pro x64
Arduino IDE: 1.8.15
ulptool: 2.4.1
binutils: binutils-esp32ulp-win32-2.28.51-esp-20191205
board: Lolin D32 Pro
esp32 core: 1.0.6

agm289, 2021-07-03:
I have got the same issue on my W10 PC and on a Ubuntu VM which are both running Arduino IDE v1.8.15.  I have a laptop running Ubuntu with v1.8.13 and that compiles without the 'undefined reference' errors.  I downgraded the Ubuntu VM to 1.8.13 and that also compiled without the errors (I couldn't fully test it as I am having problems with the VM's and my USB ports so couldn't upload).  Tomorrow I am going to try v1.8.14 on my Ubuntu VM and, see if that compiles.

I haven't got access to my PC's at the moment so cannot check on the versions of ulptool I am using.

TiborVoelcker, 2021-07-03:
Oh, so we might have the reason and a fix for the issue until it gets sorted out? 
Just downgrade to 1.8.13?

I just want to start working on it again, so that's good news. I try it out when I have the time.

gerardpalk, 2021-07-03:
Thank you for this information.  I thought I got around the issue by installing VirtualBox and running Arduino on a Ubuntu client.  I installed Arduino using the Ubuntu software installer UI, and for some reason it luckily installed v1.8.13. I had to sort through another set of issues (including USB issues), but finally got my sketch working.  However, it will be alot more convienent to run on W10, so will try the suggested solution directly on Windows.

agm289, 2021-07-04:
For your info - I installed v1.8.14 on my Ubuntu VM and got the 'undefined reference' errors so I reverted to v1.8.13 and that allowed me to compile the README example. So both my Ubuntu VM and laptop are on v1.8.13 and can compile the README example.
I then installed v1.8.13 on my W10 PC and the 'undefined reference' errors stopped but I got Python errors referenced in [this issue](https://github.com/duff2013/ulptool/issues/62). Having made the suggested changes I then got more Python errors referenced in [this issue](https://github.com/duff2013/ulptool/issues/60). So, for now, am now going to work on my Ubuntu laptop.
I don't know if the various versions of Python are causing my current problems but I will look into it later:-
Ubuntu Laptop - Python 3.8.10
Ubuntu VM - Python 3.6.9
W10 PC - Python 3.9.6

I hope this helps.

codeleger, 2022-02-10:
Maybe you are no longer looking for a solution but anyway:
@agm289 
follow @angyongen 's mention,

For me it works on WIN10 now with the following setup:

binutils-esp32ulp-win32-2.28.51-esp-20191205
python-2.7.18.amd64

modificaton as per change --> https://github.com/duff2013/ulptool/pull/80/commits/60d0f7d59439179ebf3e5e879ce4abcde37e4d3b

so add the following 3 Line in Line 19 in the platform.local.txt
`## copy '.s' (ulp) files`
`recipe.hooks.core.prebuild.01.pattern=sh -c 'cp -vp {build.source.path}/*.s {build.path}/sketch/ | sed s,^,ASMFIXUP:,'`
`recipe.hooks.core.prebuild.01.pattern.windows=cmd /c if exist "{build.source.path}\*.s" copy /y "{build.source.path}\*.s" "{build.path}\sketch\"`