duff2013/ulptool #75: error compiling assembly code

Jorge-spicymexican, 2021-06-23:
whenever I try to compile the assembly example from manual setup, I receive this error when trying to build the project.

sketch\sketch_jun22a.ino.cpp.o:(.literal._Z5setupv+0xc): undefined reference to `ulp_count'
sketch\sketch_jun22a.ino.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `ulp_entry'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.

My board is on version 1.0.0 based on the board manager. 
[image]
And the files are in the correct location according to the manual setup instructions. Any ideas what could be the problem.
Here is my images of my folders
[image]
[image]
[image]

 Any ideas what could be the issue? Thank you

Jorge-spicymexican, 2021-06-23:
i was able to fix the problem however I found out another issue the current being drawn is not the current more than 0.3 mA being shown on the tech references for the esp32. Im not if the this is working the way it is suppose too.

Bud-Spencer92, 2021-07-06:
@Jorge-spicymexican I'm facing the same problem.  Could you please tell what you have done to fix it?