duff2013/ulptool #78: Underinvestigation

outoftrun, 2021-07-12:


outoftrun, 2021-07-12:
I am running esp32 version 2.0.0-alpha1, and the latest version of your code. The esp32ulp_build_recipy.py does not find any. '*.s' files in my '/tmp/.....'  dir. If I manual copy them from my src dir everything works but whatever function had copied them to that dir is no longer doing so.

==============================================
uname -a "Linux ThinkPad-T61 5.4.0-77-lowlatency #86-Ubuntu SMP PREEMPT Thu Jun 17 03:26:36 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux"

mahesh2000, 2021-07-27:
This seems to have fixed my problem. I'm using the `ulp_README.ino` sketch from `ulp_examples`. i copied over the ulp.S file from the sketch folder to `C:\Users\<user>\AppData\Local\Temp\arduino_build_249645\sketch`. The contents of the folder are as below, with the ulp.S having been manually copied over:

```
size in bytes, filename:
      18 tmp.ulp.d
       0 tmp.ulp.lst
     444 tmp.ulp.o
       0 tmp.ulp.pS
     **313 ulp.s**
      18 ulp.ulp.d
     448 ulp.ulp.lst
     572 ulp.ulp.o
     174 ulp.ulp.pS
      36 ulp_main.bin
     492 ulp_main.bin.bin.o
     339 ulp_main.common.d
     453 ulp_main.common.ld
   8,716 ulp_main.elf
     183 ulp_main.h
     198 ulp_main.ld
   1,253 ulp_main.map
      40 ulp_main.sym
   1,641 ulp_README.ino.cpp
  15,330 ulp_README.ino.cpp.d
 763,616 ulp_README.ino.cpp.o
21 File(s)        794,284 bytes

```

mahesh2000, 2021-07-27:
Also, printing the contents of `glob` in 
`C:\Users\<user>\Documents\ArduinoData\packages\esp32\tools\ulptool\src\esp32ulp_build_recipe.py`,

```
    print("glob:")
    print(glob.glob('*.*'))
```

shows the following _without_ `ulp.S` the first time around. When run the second time, after the file has been copied over, it shows the result below:

```
glob:
['esp32_ulp_test1.ino.cpp', 'esp32_ulp_test1.ino.cpp.d', 'esp32_ulp_test1.ino.cpp.o', 'tmp.ulp.d', 'tmp.ulp.lst', 'tmp.ulp.o', 'tmp.ulp.pS', 'ulp.s', 'ulp_main.bin', 'ulp_main.bin.bin.o', 'ulp_main.common.d', 'ulp_main.common.ld', 'ulp_main.elf', 'ulp_main.h', 'ulp_main.ld', 'ulp_main.map', 'ulp_main.sym']
```

xandercrews, 2021-07-31:
i had the same errors "undefined reference to `ulp_count'" and ".. to 'ulp_entry'".

i confirm that finding the temp folder for the build (in /var/folders/*/*/*/arduino_build_*/ on macos), deleting sketch/*, copying ulp.s over manually and then recompiling works.

xandercrews, 2021-07-31:
so, when i use a hook to copy the .s file(s) to the tempdir it resolves it afaict (not windows):

```
recipe.hooks.core.prebuild.01.pattern=sh -c 'cp -vp {build.source.path}/*.s {build.path}/sketch/ | sed s,^,ASMFIXUP:,'
```

mahesh2000, 2021-07-31:
is there a windows substitute?

xandercrews, 2021-07-31:
i should think some similar xcopy command but i don't have a windows machine to try it on- in linux the key problem is that it won't invoke the command through a shell to expand the glob so you have to explicitly run it in one.

you could also make a little python program that you pass {biuld.source.path} and {build.path}, which would glob(os.path.join(sys.argv[1], '*.s')) and copy what it finds into sys.argv[2], similar to how these hooks invoke esp32ulp_build_recipe.py

xandercrews, 2021-07-31:
if python isn't your jam and you'd like help, lmk

xandercrews, 2021-07-31:
i wonder what broke.  maybe arduino gui used to copy case-insensitive *.S and now it's case-sensitive?

angyongen, 2021-08-06:
`recipe.hooks.core.prebuild.01.pattern.windows=cmd /c copy /y {build.source.path}\*.s {build.path}\sketch\` works for me (windows)

mahesh2000, 2021-08-06:
Python definitely isn't my jam, bread or butter :-). Any help is
appreciated. Right now, I'm copying over s files when updated to the temp
folder. Your help would be greatly appreciated!

angyongen, 2021-08-07:
new updated command because my previous one had an error when there are no .s files. and also doesnt work if source path has spaces
(place this in platforms.local.txt):
`recipe.hooks.core.prebuild.01.pattern.windows=cmd /c if exist "{build.source.path}\*.s" copy /y "{build.source.path}\*.s" "{build.path}\sketch\"`
@mahesh2000 maybe you can try this

mahesh2000, 2021-08-11:
that seems to work!

jacoboth7, 2022-11-14:
This solved my problem as well

bemastaff, 2024-09-23:
> so, when i use a hook to copy the .s file(s) to the tempdir it resolves it afaict (not windows):
> 
> ```
> recipe.hooks.core.prebuild.01.pattern=sh -c 'cp -vp {build.source.path}/*.s {build.path}/sketch/ | sed s,^,ASMFIXUP:,'
> ```

This absolutely saved me.  Thank you!