duff2013/ulptool #82: File path incorrect

nottledim, 2021-10-17:
Recently updated arduino to 1.8.16 and installed esp 2.0.0.  I now get this error:
```
Traceback (most recent call last):
  File "/home/dick/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 560, in <module>
    main(sys.argv[1:])
  File "/home/dick/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 103, in main
    build_ulp(PATHS, ulp_files, board_options, False)
  File "/home/dick/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 268, in build_ulp
    sdk_hash = md5(os.path.join(PATHS['core'] , 'tools', 'sdk', 'include', 'config', 'sdkconfig.h'))
  File "/home/dick/.arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 554, in md5
    with open(fname, "rb") as f:
IOError: [Errno 2] No such file or directory: '/home/dick/.arduino15/packages/esp32/hardware/esp32/2.0.0/tools/sdk/include/config/sdkconfig.h'
```
I can't work out what is issuing this error but the path it's asking for is wrong.  I think it should be:
```
hardware/esp32/2.0.0/tools/sdk/esp32/include/config/sdkconfig.h
```
Is there something I haven't updated correctly?

Snipeye, 2022-08-05:
@nottledim I don't suppose you ever got a response of any sort?  I think this was not built for ESP32-Arduino Core 2.0, but only v1.0 and such...

nottledim, 2022-08-05:
No, I bodged a few things to get it to compile but the ulp code didn't work properly.  I gave up.  I think you're probably right about Core version.