duff2013/ulptool #84: Error compiling on Mac with IDE 18.6

ChassaB, 2021-10-30:
I see the same sort of issue in several other posts but I can't see a resolution. happy to follow any instructions to delve deeper. Have triple checked files and installation directories.

Error message below.

Traceback (most recent call last):
  File "/Users/charlesburgess/Library/Arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 560, in <module>
    main(sys.argv[1:])
  File "/Users/charlesburgess/Library/Arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 103, in main
    build_ulp(PATHS, ulp_files, board_options, False)
  File "/Users/charlesburgess/Library/Arduino15/packages/esp32/tools/ulptool/src/esp32ulp_build_recipe.py", line 137, in build_ulp
    proc = subprocess.Popen(cmd[1],stdout=subprocess.PIPE,stderr=subprocess.PIPE,shell=False)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
exit status 1
Error compiling for board ESP32 Dev Module.

davidhampgonsalves, 2022-02-17:
You probably are using the incorrect binutils-esp32ulp. You want "binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz" not the s2 version("binutils-esp32s2ulp-macos-2.28.51-esp-20191205.tar.gz").

ChassaB, 2022-03-05:
Thanks I'll check if I did that and give this a go over the next week.